mne-tools/mne-python #5868: Crop-like option for TFR with `fmin` and `fmax`

mmagnuski, 2019-01-24:
I've found out during classes where I teach mne that `.crop`-like operations are usually much more intuitive for students than having to dive into the data array and index it. With tfr they usually want to, for example selecte a time range and frequency range and then average (or not) and use in further computations/visualisations. If you agree it would be useful, there are two options:

## adding `fmin` and `fmax` to `TFR.crop()`
This would allow for example for
```python
tfr.crop(tmin=0.2, tmax=0.4, fmin=5, fmax=7)
```
@larsoner mentioned that he does not like this solution very much because `.crop()` sounds more time-related and it would stay to mean "crop in time" if we instead add another method for freq-cropping (Eric, please correct me if I misrepresent). I actually like putting it all into `.crop()` since I much prefer using kwargs anyway for clarity. Also crop sounds to me equally well suited for time, space or frequency (like trim, for example, but I'm not a native so my argument is weaker here :) ). I also think it would be easier for new mne users (like my students) to digest - one method for selection instead of multiple methods depending on the dimension type.

## adding another method for frequency cropping
Another method that would work like `.crop()` but for frequency dimension. I don't have a good idea for the method name in this case but just assuming it would be called `.frop()` the example from previous section would be:
```python
tfr.crop(tmin=0.2, tmax=0.4).frop(fmin=5, fmax=7)
```
obviously I'm not suggesting to actually call it `.frop()` :)

cbrnr, 2019-01-25:
I like the first approach better, because I don't associate crop with time.

agramfort, 2019-01-25:
I am fine with first solution too

larsoner, 2019-01-29:
Thinking about it more, I'm fine with `crop`. If we think of a "crop" operation as "trimming the current raw / average / epoch / source data for all channels/sources", then you can crop in freq just as easily as time, and it's different from other forms of subselection we have, such as trimming the set of epochs with `epochs[10:20]`, subselecting source vertices with `stc.in_label`, or the set of channels with `epochs.pick_types`, etc.

massich, 2019-01-29:
I might be wrong but *cropping in frequency* with `fmin=5, fmax=7` isn't this doing a bandpass selection?

If we should be able to do that I would do `time_crop` and `frequency_crop` and remove `crop`. but this is not gonna fly.

So, when I read this issue I found it great, now I've mixed feelings.

larsoner, 2019-01-29:
> I might be wrong but cropping in frequency with fmin=5, fmax=7 isn't this doing a bandpass selection?

Yes but in principle this isn't all that different from cropping in time. You (intentionally) lose information in either case, just in different ways.

agramfort, 2019-01-29:
crop is a standard mne function name. Let's stick to this.

massich, 2019-01-29:
> crop is a standard mne function name. Let's stick to this.

That's exactly why we should not change it and having `crop` and `frequency_crop` is horrible, which takes us to `crop( tmin.. fmax)`. But when I put myself into position of doing `bla.crop+Tab` in iphython I find surprising `fmin=None, fmax=None`. That's why I was asking myself (outloud) how far was that from `tfr.crop(tmin=0.2, tmax=0.4).filter( l_freq=5, h_freq=7, type='selection')`

mmagnuski, 2019-01-29:
@massich
Do you mean that when doing for example `raw.crop(` the fmin and fmax would pop up? Can we subclass the crop mixin (if there is one) specifically for TFR? When doing `tfr.crop(`  finding fmin and fmax would be intuitive, I think.

larsoner, 2019-01-29:
Yes we should just override it in the TFR classes.

mmagnuski, 2019-01-29:
Btw I have a different intuitions for `filter` - that's an operation that requires additional processing and `crop` feels more like the only thing it does is dropping some data.

massich, 2019-01-29:
OK that's fine :)

agramfort, 2019-01-29:
> Yes we should just override it in the TFR classes.

+1

>

jona-sassenhagen, 2019-02-05:
I agree `frequency_crop` is terrible, but `crop_frequency` is not bad. Though `tfr.crop(fmin=` is also fine.

PR #6080 (2019-03-30): Added frequency cropping option with `fmin` and `fmax` to BaseTFR.crop()
#### Reference issue
Fixes #5868 

#### What does this implement/fix?
Added frequency cropping functionality to BaseTFR.crop method.

#### Additional information
Works like:

```python
tfr.crop(tmin=0.2, tmax=0.4, fmin=5, fmax=7)
```
Note that for the method, i implemented a _freq_mask() function in utils.numerics,
which is mostly similar to _time_mask() (except by the errors thrown).
Since the _freq_mask function is (for now) only needed by BaseTFR.crop(),
it could alternatively be implemented in BaseTFR.crop(), or integ [... cut]
closed 2019-03-30 by agramfort