mne-tools/mne-python #5873: Restore reference electrode

AmirDJV, 2019-01-25:
Hi, 

I'm using brain vision/product to acquire and record my data from 32 electrode EEG cap.

My problem is that when I load my data my refrence electrode (Fz) is missing. More specifically, while I manage to [add a reference electrode](https://mne-tools.github.io/dev/generated/mne.add_reference_channels.html?highlight=mne%20add_reference_channels#mne.add_reference_channels) and to plot it correctly, when I export my signal to DataFrame the values are still zero. 

I found [this mailing list](https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2017-January/003817.html) with a similar problem. However, it didn't fully help me. 

I made a jupyter notebook with a reproducible example which can be found [here](https://github.com/AmirDJV/github-question/blob/master/Restore%20reference%20electrode%20.ipynb). The sample data is [here](https://github.com/AmirDJV/github-question/blob/master/data/restoreRefElectorde.fif)

Thank you very much!

cbrnr, 2019-01-25:
Which MNE version are you using? It looks like it's not the most current one, because the average reference is just added as a projection. This means that the raw plot automatically applies the projection before plotting, but the data is not affected.

Try using the current version (0.17), then your code should work as expected. Otherwise, explicitly add `projection=False` as an argument to `set_eeg_reference`.

AmirDJV, 2019-01-25:
I explicitly added `projection=False` as an argument to `set_eeg_reference` and it solved it. 

Thank you!

closed 2019-01-25 by AmirDJV