mne-tools/mne-python #5879: Inconsistant rank estimation and API in beamformers

agramfort, 2019-01-26:
I am with @britta-wstnr and @LauriParkkonen and after debugging some issues with phantom and simulated data using beamformer code here a list of things that we think should be addressed:

- `rank` default parameter is not consistent between `make_lcmv` and `make_dics`
- `rank='full'` in make_lcmv is a misnomer and it's assuming full rank if a noise_cov parameter is passed. We suggest to have an 'auto' mode that would fall back to using the `info` to infer the data rank.

We propose to create a unifying function call

`rank = compute_rank(inst, noise_cov=None, rank='auto', info=None)`

where inst can be Raw or Epochs or Covariance (info needed if Covariance.
This function would be called early in inverse pipeline so we pass explicitly
and int to all subsequent functions.

Started in #5876

agramfort, 2019-12-28:
to close @larsoner ?

larsoner, 2019-12-28:
We still haven't dealt with the XXX's having to do with tolerance levels, but we can close and get to it someday if we need to

closed 2019-12-28 by larsoner