mne-tools/mne-python #5880: (Vector) beamformer should throw error with sphere headmodel

britta-wstnr, 2019-01-26:
Leadfield rank reduction is not implemented yet for vector beamformers, so they should throw an error with leadfields that are not full rank (e.g. with sphere head models). 
Furthermore, the scalar beamformer does only complain about leadfield rank if the inversion fails. It would in general be good to inform the user about reducing the rank when having sphere head model upfront.

britta-wstnr, 2019-04-26:
Add an ``auto`` option to the ``reduce_rank`` parameter: reduce the rank of the leadfield on a voxel by voxel basis to deal with sphere models as well as unstable voxels in other headmodels. --- Think again, whether it is a good idea to make voxels behave differently.

agramfort, 2019-12-28:
@britta-wstnr @larsoner has this been addressed already? I have the feeling that maybe...

PR #7168 (2020-01-27): MRG, ENH: Extend reduce_rank=True support and better deficiency check
Closes #5880.

@agramfort `reduce_rank=True` was still only implemented for `pick_ori == 'max_power' and weight_norm is not None`. This adds support for the other modes, plus a sanity check on the gain matrix directly rather than relying on `linalg` functions to fail.

@britta-wstnr @wmvanvliet it would be good if you could take a look.
closed 2020-01-27 by larsoner