mne-tools/mne-python #5884: Problems with installation instructions

biancasama, 2019-01-28:
I was following the installation guides for MacOS and then at [point 2](https://martinos.org/mne/stable/install_mne_python.html#get-mne-and-its-dependencies) I get the following error:

```sh
(base) Biancas-MacBook-Pro:~ bt_neurospin$ curl -0 https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
name: mne
channels:
- defaults
dependencies:
- python>=3.6
- pip
- mkl
- numpy
- scipy
- matplotlib
- cython
- pyqt>=5.9
- vtk>=8
- pandas
- xlrd
- scikit-learn
- h5py
- pillow
- statsmodels
- jupyter
- pytest
- pytest-cov
- pytest-mock
- pytest-timeout
- joblib
- psutil
- numpydoc
- flake8
- spyder
- numexpr
- traits>=4.6.0
- pyface>=6
- traitsui>=6
- testpath<0.4
- pip:
  - mne
  - https://api.github.com/repos/enthought/mayavi/zipball/b3fc35218dda9776d8f1a407663bfb49783dca12
  - PySurfer[save_movie]
  - dipy --only-binary dipy
  - nibabel
  - nilearn
  - neo
  - pytest-faulthandler
  - pydocstyle
  - codespell
  - python-picard
(base) Biancas-MacBook-Pro:~ bt_neurospin$ conda env create -f environment.yml

EnvironmentFileNotFound: '/Users/bt_neurospin/environment.yml' file not found

(base) Biancas-MacBook-Pro:~ bt_neurospin$ 
```

massich, 2019-01-28:
The problem is a typo in your command. It should be `curl -O blabla` (minus capital '0') not `curl -0` (minus 'zero'). The problem is that `conda env create -f` requires the `environment.yml` file and this one is not there because `curl -0` does not create the file but rather spits out the file in the console (thats why you see all those package names).

Just make sure that you copy paste this:
```sh
curl -O https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
conda env create -f environment.yml
conda activate mne
```

massich, 2019-01-28:
Maybe we should update our documentaiton to make sure this does not happen. Maybe we should modify our sinpped so that it can be copy pasted without any problem. 

The other option is to pass the curl to conda, so that the file is not even saved in the disc.

agramfort, 2019-01-28:
we could use something like this:

https://predictablynoisy.com/sphinx-copy-buttons
https://clipboardjs.com/

>

ValMand, 2019-01-29:
This happen to me as well 
[image]

agramfort, 2019-01-29:
we definitely need to improve this. Ie make it easier to copy-paste
and/or add a warning note.

PR #5888 (2019-01-30): MRG: Fix instruction copy-paste
Take care of a number of paper cuts:

1. Change OSX instructions to use explicit `curl --remote-name ` instead of `curl -O`
2. Set CSS style of console separators in Pygments highlighting to be non-selectable, so copy-paste is easier. This is not done for Python because we have the `>>>` on the right to hide the separators.
3. Update entries in example outputs to reflect Python 3.7.
4. Update mayavi pin in environment.yml.
5. Fix a couple of minor bugs with linking:
    ```
    /home/larsoner/python/mne-python/doc/auto_tutorials/plot_object_source_estimate.rst:2 [... cut]
closed 2019-01-30 by agramfort