mne-tools/mne-python #5885: stim channels not picked correctly for BTi data

jasmainak, 2019-01-28:
Detailed instructions on how to file a bug can be found in our [FAQ](https://martinos.org/mne/stable/faq.html#i-think-i-found-a-bug-what-do-i-do).

If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the bug
When I try to pick the stim channel in BTi data, it does not do it correctly and picks almost all the channels. Here is the code to reproduce:

```Python
import mne
import os.path as op

base_path = op.join(op.dirname(mne.__file__), 'io')

data_path = op.join(base_path, 'bti', 'tests', 'data')
raw_fname = op.join(data_path, 'test_pdf_linux')
config_fname = op.join(data_path, 'test_config_linux')
headshape_fname = op.join(data_path, 'test_hs_linux')

raw = mne.io.read_raw_bti(raw_fname, config_fname=config_fname,
                          head_shape_fname=headshape_fname)
picks_stim = mne.pick_types(raw.info, stim=True)
print('Number of stim channels is: %d' % len(picks_stim))

```

#### Expected results
2

#### Actual results
249

dengemann, 2019-01-28:
No idea. What BTI data is this @jasmainak (which system + config) ?

jasmainak, 2019-01-28:
never mind, false alarm. I think I should set `meg=False` :)

closed 2019-01-28 by jasmainak