mne-tools/mne-python #5893: Segfault when running make_report

massich, 2019-01-30:
Can anyone replicate this error?

```
~/code/mne-python neurospin_2*
(mne) ❯ ipython -i ./examples/visualization/make_report.py
Python 3.6.6 |Anaconda, Inc.| (default, Oct  9 2018, 12:34:16) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.1.1 -- An enhanced Interactive Python. Type '?' for help.
zsh: segmentation fault (core dumped)  ipython -i ./examples/visualization/make_report.py

~/code/mne-python neurospin_2*
(mne) ❯ 
```

larsoner, 2019-01-30:
It is very slow but works for me.

I can almost guarantee if you add a `import faulthandler; faulthandler.enable()` at the top of `mne/report.py`, the segfault with be with Mayavi/VTK.

massich, 2019-01-30:
```
~/code/mne-python report_segfault*
(mne) ❯ ipython examples/visualization/make_report.py         
Fatal Python error: Segmentation fault

Thread 0x00007f6211648700 (most recent call first):
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/threading.py", line 295 in wait
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/threading.py", line 551 in wait
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/core/history.py", line 829 in run
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "<decorator-gen-24>", line 2 in run
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f621490a740 (most recent call first):
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 595 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "/home/sik/code/mne-python/mne/report.py", line 150 in _iterate_trans_views
  File "/home/sik/code/mne-python/mne/report.py", line 1965 in _render_trans
  File "/home/sik/code/mne-python/mne/report.py", line 298 in _iterate_files
  File "/home/sik/code/mne-python/mne/report.py", line 1470 in <genexpr>
  File "/home/sik/code/mne-python/mne/report.py", line 1470 in parse_folder
  File "/home/sik/code/mne-python/mne/utils/_logging.py", line 74 in verbose
  File "<string>", line 2 in parse_folder
  File "/home/sik/code/mne-python/examples/visualization/make_report.py", line 30 in <module>
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/utils/py3compat.py", line 188 in execfile
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/core/interactiveshell.py", line 2683 in safe_execfile
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/core/shellapp.py", line 323 in _exec_file
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/core/shellapp.py", line 391 in _run_cmd_line_code
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/core/shellapp.py", line 271 in init_code
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/terminal/ipapp.py", line 323 in initialize
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/traitlets/config/application.py", line 87 in catch_config_error
  File "<decorator-gen-112>", line 2 in initialize
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/traitlets/config/application.py", line 657 in launch_instance
  File "/home/sik/miniconda3/envs/mne/lib/python3.6/site-packages/IPython/__init__.py", line 125 in start_ipython
  File "/home/sik/miniconda3/envs/mne/bin/ipython", line 11 in <module>
zsh: segmentation fault (core dumped)  ipython examples/visualization/make_report.py
```

larsoner, 2019-01-30:
One thing you can try is add this before any `mlab.screenshot()` call in `report.py`:
```
try:
    mlab.process_ui_events()
except AttributeError:
    pass
```

massich, 2019-01-30:
I get the same

GuillaumeFavelier, 2019-01-31:
but then I tried on my `(mne-vispy)` environment and then I got the following :
```
(mne-vispy) ➜  mne-python git:(master) ✗ ipython -i ./examples/visualization/make_report.py
Python 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.
Embedding : jquery-1.10.2.min.js
Embedding : jquery-ui.min.js
Embedding : bootstrap.min.js
Embedding : jquery-ui.min.css
Embedding : bootstrap.min.css
Iterating over 28 potential files (this may take some time)
Rendering : /home/guillaume/mne_data/MNE-sample-data/MEG/sample/all-trans.fif
Using outer_skin.surf for head surface.
Getting helmet for system 306m
[1]    15019 segmentation fault (core dumped)  ipython -i ./examples/visualization/make_report.py
```

I don't know if it's related because in my case the script start at least.

larsoner, 2019-09-21:
This was hopefully fixed by #6434

closed 2019-09-21 by larsoner