mne-tools/mne-python #5903: Use labels for WIP and MRG

cbrnr, 2019-02-07:
Instead of prepending [WIP] or [MRG] to PR titles, we could create WIP and MRG labels that let the author indicate the current status of a PR. One benefit of this would be the ability to filter through WIP and MRG much more easily. In addition, titles get shorter and thus easier to read. WDYT?

agramfort, 2019-02-07:
no strong feeling. It's just my old habits. I also see by doing this the
status in the title of the email notification which I find quite handy

>

larsoner, 2019-02-07:
> let the author indicate the current status of a PR

AFAIK only people with write permissions on the repo have control over labels, everyone can control their own titles.

closed 2019-02-07 by larsoner

cbrnr, 2019-02-07:
You're right, this is something that GitHub could add.

cbrnr, 2019-02-14:
Wow - https://github.blog/2019-02-14-introducing-draft-pull-requests/

Let's use that instead!

larsoner, 2019-02-15:
Unfortunately I just tried it, and only LGTM and AppVeyor ran, CircleCI and Travis did not. So we might have to wait for the CIs to adjust to this new system

larsoner, 2019-02-15:
... actually it might have just been very (spuriously) delayed. I'll try on my next relevant PR