mne-tools/mne-python #5907: Need a way to show/hide annotations in raw.plot

cbrnr, 2019-02-08:
With the latest changes in how MNE handles discrete events (it now creates annotations), manually adding annotations in `raw.plot` can get impossible. For example:

```
import mne
from mne.datasets import eegbci

fpath = eegbci.load_data(subject=1, runs=4)
raw = mne.io.read_raw_edf(fpath[0], preload=True)
raw.plot()
```

This results in the following raw browser window:

[image]

Annotations present in the EDF+ file fill up the whole signal. If I want to create new annotations (called e.g. "BAD" to mark artifact segments), I cannot create new annotations by dragging with my mouse, because this apparently requires me to start in an empty portion (i.e. a portion without any annotation).  Even though I somehow managed to add a new annotation segment once, I think we need some way to show/hide existing annotations. Maybe the current annotations dialog window can be used, which already contains a list of annotations present in the file.

Furthermore, the current position as indicated by a gray rectangle inside the navigation bar at the bottom of the window is barely visible because of the annotations. Maybe darkening the rectangle would solve this (minor) problem.

agramfort, 2019-02-08:
good point. Yes we need offer a way to toggle certain annotations.

>

cbrnr, 2019-06-05:
I think I found a way to fix this problem plus also fix #4769: we simply need to disable the current auto-snap to annotation start/end behavior. I somehow managed to do this once, but I don't know what exactly I changed. @larsoner do you know a quick way to disable the behavior? If disabled, I can create annotations within other annotations, which means that there is no need to show/hide annotations anymore.

I think this will be a useful change, because annotations of the same type are still merged (just drag one into the other and they will get merged), but you can create different annotations within other annotations.

larsoner, 2019-06-05:
No idea, you'll have to dig into the code :(

cbrnr, 2019-06-05:
OK, I've already started anyway, but was hoping that you'd have a fresh memory of this rather convoluted code. Anyway, assuming that I'll be able to disable this behavior, would you be +1 on that change?

larsoner, 2019-06-05:
Maybe holding some modifier key (control?) could disable it?

cbrnr, 2019-06-05:
Yes, good idea. Maybe Shift+Click would work best. We could also do it the other way around, don't snap by default and snap with the modifier key (because I believe snapping is rarely needed).

cbrnr, 2019-06-06:
I found the code responsible for editing existing annotations: https://github.com/mne-tools/mne-python/blob/0679e350cc20a07f3d043fec0be9a2153aa99a2b/mne/viz/utils.py#L2039

The callback is assigned here: https://github.com/mne-tools/mne-python/blob/0679e350cc20a07f3d043fec0be9a2153aa99a2b/mne/viz/utils.py#L946

Now if I remove this callback assignment (or change the `_on_hover` function so that it does nothing), I get the desired behavior (which means I can create annotations over existing annotations, and if they are of the same type they get merged). I cannot edit existing annotations though (other than removing completely by right-clicking and extending by adding a new annotation from outside, but making an existing segment smaller doesn't work anymore).

I don't need the edit on hover behavior, because I either delete an annotation and create a new one, or I can still extend by merging a new annotation. But ideally, holding down the Shift key should trigger the old behavior. The only problem is that I don't know how to do that - the `_on_hover` function would have to constantly check if the Shift key is pressed or not.

larsoner, 2019-06-06:
> But ideally, holding down the Shift key should trigger the old behavior.

I would do it the other way, keep the current behavior by default and disable it if shift is pressed. It's less disruptive in case people are currently using the snapping.

>  The only problem is that I don't know how to do that - the _on_hover function would have to constantly check if the Shift key is pressed or not.

Something like this should do it

https://stackoverflow.com/questions/18141069/matplotlib-how-to-pick-up-shift-click-on-figure

PR #6430 (2019-06-13): Modifier key has to be pressed for snapping to annotations
Fixes #5907. Currently, I had to set the modifier key to 'v', because 'shift' or 'control' don't work. @larsoner I think I don't even need to manually check for key press and release events as you suggested, because a [`MouseEvent`](https://matplotlib.org/api/backend_bases_api.html#matplotlib.backend_bases.MouseEvent) also has a `key` attribute (seems to be new).

WDYT? Do you have any idea why it doesn't work with shift? Also, I really would like to have the snapping behavior mapped to the modifier (I received some feedback from peopl [... cut]
closed 2019-06-13 by larsoner