mne-tools/mne-python #5918: Plotting a modified ICA object while awaiting arguments

IDCtolik, 2019-02-11:
Hello,

I'm trying to get some plots of an ICA object, then to be prompted to enter channels to exclude upon examining my plots, and finally have my script print the plots of the now modified ICA object.
The issue is that none of my plots are complete (blank white windows) until I enter the missing arguments...
Any suggestions for a workaround?

agramfort, 2019-02-11:
Sorry I don't understand your problem. Can you clarify. What are you trying
to do?

>

IDCtolik, 2019-02-11:
Hi, thank you for your reply!

so this is the order of events:

*create a new ICA , lets denote it as ica0 object
*run a few plot functions on the ica0
*examine the plots (the first set of plots should be complete at this point, before the script blocks for waiting for my input)
*prompt the user for input, the input should be indices of channels to exclude
*exclude the channels from ica0
*call plot functions on the updated ica0

The issue is that when the script blocks upon reaching the "arguments from user" state my first set of plots does not complete

agramfort, 2019-02-11:
what do you mean the script blocks? do you set block=True in matplotlib?

>

IDCtolik, 2019-02-11:
I mean the script  will do nothing until I give the user input, I didn't set block=true in matplotlib, will adding a picture, code snippet help?

agramfort, 2019-02-11:
user input?

>

IDCtolik, 2019-02-22:
Sorry for the long delay, I solved the issue that I'll just leave it here for future reference:

My issue was with the following part of code:
`
epochs.plot()
# ICA
ica = ICA(n_components=0.95, method='fastica', random_state=1).fit(epochs)
##Plot components and sources
plotHelper.fileOpener()
##Exclude the eye movement component from data 
ICAtoExclude = input("Which componenets should be excluded? ")
ICAtoExclude = [int(x) for x in ICAtoExclude.split(',')]
ica = ica.apply(epochs1, exclude=ICAtoExclude)
ica.plot()
`
The intention was to plot epochs, and the user of the script decides on the input for "ICAtoExclude" based on that. What happened in practice is that the plot simply won't load.

I tried to add a line after my epochs plot:
`plt.show(block=true)`
*im using the interactive mode

It didn't work for me, I got the plot but the program didn't resume upon closing it.

What eventually worked is adding:
`plt.pause(1)`
instead of show()
*note that this is a messy solution, it worked when I set the pause for 1 but didn't when I tried 0.1, so if I understand it correctly it's hardware dependent(?)

agramfort, 2019-02-23:
what is this ?

##Plot components and sources
plotHelper.fileOpener()

IDCtolik, 2019-02-23:
Thank you, it's irrelevant to the issue, I overlooked it when I was cleaning redundancy before posting here

agramfort, 2019-02-23:
i don't know how to replicate and what to fix here. I would suspect interactions between matplotlib event loop and ipython or jupyter or spyder. I don't know what environment you use.

ok to close?

closed 2019-12-28 by agramfort