mne-tools/mne-python #5924: Feature: Source labels simulation

makkostya, 2019-02-12:
As I said in #5058 , I start the discussion of source labels simulation feature, so feel free to participate in it.

The objective is to create a function which would provide a simple an general way to simulate cortical regions which would be used for stc simulation.

There is already a function `grow_labels` in MNE which creates growing regions around a seed point but it requires some extra scripting from user that, I think, can be included to API.
For a moment, here is a possible API of such a function.

```
def simulate_source_labels(subject, parcels, seeds, extents,
                           overlap=True, grow_outside_parcel=True, subjects_dir=None,
                           n_jobs=1, names=None,random_state=None, surface='white'):
    """Simulate source labels.
    
    Parameters
    ----------
    subject : string
        Name of the subject as in SUBJECTS_DIR.
    parcels : instance of Label, str | list
        Define where the source seed will be chosen. If str, can be
        'rh' or 'lh'.
    seeds : int, str | list
        Seed, or list of seeds. If int, seed represents a vertex number in
        corresponding parcel. If str, can be either 'random' or 'barycenter'.
    extents : float | list
        Extents (radius in mm) of the labels, i.e. maximum geodesic distance 
        on the white matter surface from the seed. If 0, source contains only one
        seed vertex.
    overlap : bool
        Produce overlapping labels. If True (default), the resulting labels
        can be overlapping. If False, each label will be grown one step at a
        time, and occupied territory will not be invaded.
    grow_outside_parcel : bool | list
        Let regions grow outside the parcel where seed was defined.
    subjects_dir : string
        Path to SUBJECTS_DIR if not set in the environment.
    n_jobs : int
        Number of jobs to run in parallel.
    names : None | list of str
        Assign names to the new labels (list needs to have the same length as
        seeds).
    random_state : None | int | np.random.RandomState
        To specify the random generator state.
    surface : string
        The surface used to grow the labels, defaults to the white surface.
    
    Returns
    -------
    labels : list of Label
    """
```
Usage example:

```
subject = 'sample'
labels = read_labels_from_annot(subject)

##################################################3
# Simulate 3 source regions with different parameters:
# 1. Select random seed inside "bankssts-lh" parcel, grow region of radius 10mm
#    which can potentially grow outside the initial parcel.
# 2. Select barycenter of "bankssts-rh" parcel, grow region of radius 15mm 
#    and constrain the region to lie completely inside the initial parcel.
# 3. Select 42th vertice of right hemi surface mesh and keep it as a single source.

parcels = [labels[0], labels[1], 'rh']
seeds = ['random', 'barycenter', 42]
extents = [10., 15, 0]
gop=[True, False, True]
source_labels = simulate_source_labels(subject, parcels, seeds, extents, 
                                       grow_outside_parcel=gop)
```

Another important question to discuss is adding visual interface which would allow manual selection of sources on the cortical mesh. @christospapa can provide some code which already exists.

massich, 2019-02-12:
@makkostya could you open a PR with the code you are proposing and reference #5058? Ping me if you need any help with it.

makkostya, 2019-02-12:
For the moment there is no code to pull. I just started the discussion.

agramfort, 2019-02-12:
I would go with:

```
def simulate_label(subject, label, seeds, extent=0.,
                                grow_outside=True, subjects_dir=None,
                                name=None, random_state=None,
                                surface='white'):
    """Simulate source labels.
    
    Parameters
    ----------
    subject : string
        Name of the subject as in SUBJECTS_DIR.
    label : instance of Label | str
        Define where the source seed will be chosen. If str, can be
        'rh' or 'lh'.
    location : 'random' | 'barycenter' | int
        Location to grow label from. If int, location represents the vertex number
        in corresponding label. If str, can be either 'random' or 'barycenter'.
    extent : float
        Extents (radius in mm) of the labels, i.e. maximum geodesic distance 
        on the white matter surface from the seed. If 0, source contains only one
        seed vertex.
    grow_outside : bool
        Let labels grow outside the original label where location was defined.
    subjects_dir : string
        Path to SUBJECTS_DIR if not set in the environment.
    names : None | str
        Assign name to the new label.
    random_state : None | int | np.random.RandomState
        To specify the random generator state.
    surface : string
        The surface used to simulated the label, defaults to the white surface.
  
    Returns
    -------
    label : instance of Label
            The simulated label.
    """
```

then if you want many labels just use a for loop.

wdyt?

makkostya, 2019-02-13:
I put list of labels as possible input because theoretically user could want to simulate two regions issued from different parcels, allowed to grow outside their parcels but not allow these regions to intersect. In this case two parcels could not be treated independently via for loop. Even though I think that this scenario is quite unlikely.

Another advantage of giving list of labels as an input is possibility of using "n_jobs" parameter of `grow_labels`  function which will be called inside `simulate_labels` for parallel computing.

agramfort, 2019-02-13:
let's do one thing at a time and add options when need emerges. I
think that n_jobs here is also overkill given how fast it should be to
compute.

larsoner, 2019-04-24:
Fixed by #6193

PR #6193 (2019-04-24): MRG: add convenience function to select sources for simulation
To simulate cortical activity, the user must select a region of the cortex to activate. This can be done with `grow_labels`, but requires knowing the absolute index of the vertices.

This commit adds a new function select_sources which allows a user to create a region based on a pre-existing label object. The idea is to make this function the defacto way to select sources for simulations.

Closes #5924
closed 2019-04-24 by larsoner