mne-tools/mne-python #5926: 'module' object has no attribute 'fiff'

jeon11, 2019-02-12:
Hello, I am not sure if this would be the appropriate place to ask a question, but I am new to exploring the mne, and I used pip to install mne. 
`pip install men --user`  

I tried to follow along the tutorial to get a sense of the toolbox, but I faced an error with 
```
'module' object has no attribute 'fiff'
```

I was wondering if any other people had similar problem as me. Thanks!

larsoner, 2019-02-12:
Is that actually the line you used? `mne` is misspelled.

What version did `pip` say it installed? What's the complete traceback?

I have not seen an error like this before.

jeon11, 2019-02-12:
My apologies. I used `mne'. The `men` above was auto-correction! 

But, yes, I used `pip install mne --user` and it now shows: 
`Requirement already satisfied: mne in ./Library/Python/2.7/lib/python/site-packages (0.17.0)`

my pip version is 19.0.1. It does say I can upgrade to 19.0.2. I'll try updating it and reinstalling again meanwhile.

agramfort, 2019-02-12:
what is the line that created the crash?

did you by any chance name a file mne.py?

>

jeon11, 2019-02-12:
Nope, I didn't have the file named mne.py.
I called in: 
`raw = mne.fiff.Raw(raw_fname)` (same as the tutorial). Then it showed me:

```
AttributeErrorTraceback (most recent call last)
<ipython-input-7-fac27c570de0> in <module>()
----> 1 raw = mne.fiff.Raw(raw_fname)

AttributeError: 'module' object has no attribute 'fiff'
```

However, I'm thinking the mne didn't properly get installed on my computer, because suddenly when I tried to `pip install --upgrade pip --user`, it suddenly showed:
```
Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '/Users/Jin/Library/Python/2.7/lib/python/site-packages/pip-19.0.1.dist-info/METADATA'
``` 
Trying to look into this problem, but if you have any pointers to this problem, that would be grateful! 

Also, posted my question on Stack Overflow with bit of more details here:
https://stackoverflow.com/questions/54658517/environmenterror-no-such-file-or-directory-metadata

agramfort, 2019-02-12:
you're code is too old. I don't know where you got this code snippet. mne
has changed since then.

>

jeon11, 2019-02-12:
Hmm I used `pip freeze` and it showed `mne==0.17.0` which I am aware it as the latest version support for Python 2. 

Are you referring to the `raw = mne.fiff.Raw(raw_fname)` that is being old, or the error traceback message that's old?

agramfort, 2019-02-12:
mne.fiff is not the right syntax for mne 0.17

Please see current documentation 

It should me mne.io now

jeon11, 2019-02-12:
Oh I guess I was looking at the outdated tutorials page? 

I was trying to follow along https://mne-tools.github.io/mne-python-intro/#access-raw-data

Should I refer to http://martinos.org/mne/stable/tutorials/philosophy.html as a good starting point?

agramfort, 2019-02-13:
yes it's outdated.

@larsoner @drammock  what do you think we do about this? I am not sure we have an equivalent 10mins  tutorial. I like the idea of having getting started with MNE in 60 mins just after the link that says how to install stuff.

See for example the https://pytorch.org/tutorials/ that first points to https://pytorch.org/tutorials/beginner/deep_learning_60min_blitz.html

drammock, 2019-02-13:
@jeon11 the more up-to-date tutorials are here: http://martinos.org/mne/stable/auto_tutorials/index.html

In particular, this one covers similar ground to what you were looking at: martinos.org/mne/stable/auto_tutorials/plot_sensors_time_frequency.html

drammock, 2019-02-13:
@agramfort I've been wondering about that page...  came across it a couple days ago.  Was planning to delete it as part of my next PR.

I like the "60 minutes" idea, I'll keep that in mind when putting together the new tutorial sequence.  Let's continue discussion in #5937.

larsoner, 2019-02-13:
Sounds like a plan to me

closed 2019-02-14 by agramfort