mne-tools/mne-python #5928: [BUG] Computation of noise in NAI beamformer

britta-wstnr, 2019-02-13:
The noise computation in the NAI beamformer is flawed. In case of full rank matrices, it throws the following error now (as discovered by @MadsJensen):

```
'Cannot compute noise subspace with a full-rank  covariance matrix and no 
regularization. Try  manually specifying the rank of the covariance matrix or 
using regularization.'
```
However, in this case, the beamformer should just rely on the smallest eigenvalue of the covariance matrix SVD, which represents the noise level.

@agramfort @wmvanvliet @sarangnemo

agramfort, 2019-02-13:
maybe we can look at this after the completion of the rank refactoring ? I would suggest to address this during the sprint too.

sarangnemo, 2019-02-13:
@agramfort not sure of the timeline of the rank refactoring, but this bug is a blocker for us locally, we have several users affected.

larsoner, 2019-02-13:
I can take a look to see if the fix is simple enough. This can become one more thing to consider when unifying later.

agramfort, 2019-02-13:
ok fair enough

can we have a code snippet to replicate / fix?

larsoner, 2019-02-13:
@sarangnemo @britta-wstnr I tried changing a test in `test_lcmv.py` that computes:
```
filters = make_lcmv(evoked.info, fwd, data_cov, reg=0.,
                    noise_cov=noise_cov, pick_ori='max-power',
                    weight_norm='nai')
```
But I did not hit this error. I also had to remove the three active projectors. That's because this failure mode occurs when:

1. `reg=0` (which causes `loading_factor=0` to be returned by `_reg_pinv`), *and*
2. `rank >= len(Cm)`

Perhaps this conditional needs to be `rank > len(Cm)`? I'll implement it and modify the tests unless someone knows it's a bad idea.

larsoner, 2019-02-13:
https://github.com/mne-tools/mne-python/pull/5934

PR #5934 (2019-02-14): BUG: Fix beamformer bug with full rank and reg=0.
Closes #5928.

@britta-wstnr can you see if the fix looks reasonable and it fixes your problem?
closed 2019-02-14 by agramfort