mne-tools/mne-python #5929: Are martinos.org and mne-tools.github.io in sync?

massich, 2019-02-13:
[martinos](https://martinos.org/mne/dev/documentation.html#collapse_io) and [github](https://mne-tools.github.io/dev/documentation.html#collapse_io) do not show the same. Suggesting that the dev documentation in martions website might be out of date.

We could:
(a) redirect the dev doc to github.io (I'm not sure about the implcations of this)
or (b) use automations to push the website to martinos every time that master updates. This has the advantage that once set we could forget about it.

larsoner, 2019-02-13:
@agramfort has to do it manually due to restricted access

larsoner, 2019-02-13:
(and we want to keep the doc on the Martino's site)

closed 2019-02-13 by larsoner