mne-tools/mne-python #5933: Bug: mne import failure

marsipu, 2019-02-13:
Upgrading a stable version of mne-python (0.17.0) to the current developement version (0.18.dev0) leads to an error when importing the module mne. The upgrade was done via the pip install --upgrade command.

OS: Windows 10
conda --version: conda 4.6.3
python --version: Python 3.6.5 :: Anaconda, Inc.

# Creating a new mne environment with conda
```
(Anaconda Prompt)
 curl --remote-name https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
 conda env create -f environment.yml -n mne_dev

```
# pip-upgrade to developement version
```
(Anaconda Prompt)
conda activate mne_dev
pip install --upgrade --no-deps git+https://github.com/mne-tools/mne-python.git
```

# Error
```
(Anaconda Prompt)
python
>>> import mne
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda3\envs\mne_dev\lib\site-packages\mne\__init__.py", line 25, in <module>
    from .io.pick import (pick_types, pick_channels,
  File "C:\Anaconda3\envs\mne_dev\lib\site-packages\mne\io\__init__.py", line 24, in <module>
    from . import edf
  File "C:\Anaconda3\envs\mne_dev\lib\site-packages\mne\io\edf\__init__.py", line 7, in <module>
    from .edf import read_raw_edf, read_raw_bdf, read_raw_gdf, find_edf_events
  File "C:\Anaconda3\envs\mne_dev\lib\site-packages\mne\io\edf\edf.py", line 34, in <module>
    md5='12134a9be7e0bfa5941e95f8bfd330f7')
  File "C:\Anaconda3\envs\mne_dev\lib\site-packages\mne\io\edf\_utils.py", line 19, in _load_gdf_events_lut
    fname)
ValueError: File C:\Anaconda3\envs\mne_dev\lib\site-packages\mne\io\edf\gdf_encodes.txt is corrupted. mdf5 hashes don't match.
```

**Another try with a cloned git repository and "python setup.py develop" resulted in the same error.**

#### Expected results
A working mne-python environment

#### Actual results
Import of the module "mne" not possible

#### Additional information
mne.sys_info() not available due to import failure of mne

agramfort, 2019-02-14:
I cannot replicate the problem on my mac.

maybe @OlehKSS you can try to replicate on your windows machine?

OlehKSS, 2019-02-14:
@agramfort I replicated the steps mentioned above and got the same issue on my Windows 10 laptop

massich, 2019-02-14:
does it mean that appveyor is broken?

OlehKSS, 2019-02-14:
This is what I get
41696b162526f559dd7d139352eae178

agramfort, 2019-02-14:
can you compare the files you get on windows and on linux? can you confirm that the same file produces differents hashes on unix or win?

OlehKSS, 2019-02-14:
This is the file that I have on my system
[gdf_encodes.txt](https://github.com/mne-tools/mne-python/files/2864976/gdf_encodes.txt)

agramfort, 2019-02-14:
can you check that you have different hash on windows or linux?

>

larsoner, 2019-02-14:
The files have different line endings, on Windows they are CR-LF and on Unix LF

larsoner, 2019-02-14:
(yes the hashes are different, the Windows one is 41696b162526f559dd7d139352eae178 and Unix 12134a9be7e0bfa5941e95f8bfd330f7)

PR #5942 (2019-02-14): BUG: Add CR-LF hash
Closes #5933.

We should backport this before releasing 0.17.1.
closed 2019-02-14 by massich