mne-tools/mne-python #5935: Default MNE_DATASET_PATH variable in config

hubertjb, 2019-02-13:
I am working with MNE's data fetchers, and more specifically looking at using the [Sleep Physionet dataset](https://mne-tools.github.io/dev/generated/mne.datasets.sleep_physionet.age.fetch_data.html#mne.datasets.sleep_physionet.age.fetch_data). I would like to make sure that when I download the data for the first time it gets saved to a specific directory on my machine.

I saw that the argument `path` of `mne.datasets.sleep_physionet.age.fetch_data` lets the user specify where to store the data. If it's not provided, then the environment variable/config parameter `MNE_DATASETS_PHYSIONET_SLEEP_PATH` is used instead; if that also doesn't exist, then the data is stored under `~/mne_data`.

Would there be any interest in adding an environment variable for the default storage path? E.g., `'MNE_DATASETS_PATH': '~/mne_data'`.

Thanks!

cc: @massich

larsoner, 2019-02-13:
It already exists, the docstring is probably just out of date. Can you check?

larsoner, 2019-02-13:
(The key is `MNE_DATA`)

hubertjb, 2019-02-13:
Aah got it thanks!

So it appears here: https://github.com/mne-tools/mne-python/blob/2502260f48b84299c00530ab09fdc9e2b2a4cddf/mne/datasets/utils.py#L170
and here:
https://github.com/mne-tools/mne-python/blob/2502260f48b84299c00530ab09fdc9e2b2a4cddf/mne/utils/config.py#L89

So all that's missing is the information in the docstrings of the `fetch_data` functions? If so I can update them.

larsoner, 2019-02-13:
> If so I can update them.

It's probably best to start using the `doccer` methods we have available now. See how we do it here:

https://github.com/mne-tools/mne-python/pull/4511/files#diff-c0cc0663d36fcb3b69d6e33a22573db1

Basically you separate what is always said the same way into chunks, then construct the docstring using `%(whatever)s`. Then you need to decorate the function with either `@verbose` (if it has a verbose kwarg) or `@fill_doc` if it does not.

larsoner, 2023-10-01:
We document `MNE_DATA` nowadays so we can close

closed 2023-10-01 by larsoner