mne-tools/mne-python #5943: Lodaing BDF file fails on UTF-8 decoding error

danieldiamont, 2019-02-14:
#### Describe the bug
Please provide a clear and concise description of the bug.

The header of my BDF file has a patient with a name that contains the character `è`, which causes line 600 of `io/edf/edf.py` to fail with error:

```python
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe8 in position 4: invalid continuation byte
```

#### Steps and/or code to reproduce

Loading any `.bdf` file with that accent causes the issue. The issue occurs with the most recent pip package and the current head of master.

#### Additional information

We temporarily solved our issue by changing the line to:

```python
# patient ID
pid = fid.read(80).decode(errors='ignore')
```

However, it may be worthwhile to replace the offending character with a space or something harmless.

larsoner, 2019-02-14:
This suggests that EDF does not encode using UTF-8. We should find out what encoding they use, and `decode()` with that.

agramfort, 2019-02-14:
how did you write the edf file? is utf-8 accepted by the standard? is there
a way to know if the edf file should be decoded in latin1 or utf-8?

>

cbrnr, 2019-02-15:
AFAIK the standard does not mention which encoding should be used. We've stumbled into a similar issue for GDF files, which we solved by using `.decode('latin-1')`. This should take care of the accent (and other special characters).

agramfort, 2019-02-15:
cool. Who sends the PR and shares a test file?

>

cbrnr, 2019-02-15:
I can do the PR if nobody else beats me to it.

agramfort, 2019-02-15:
go for it please

PR #5949 (2019-02-16): [MRG] Use latin-1 encoding for some EDF header fields
Fixes #5943.
closed 2019-02-16 by agramfort