mne-tools/mne-python #5945: max-power beamformer doesn't compute power and crashes

jaaval, 2019-02-15:
#### Describe the bug
make_lcmv doesn't check the validity of weight_norm parameter.

#### Steps and/or code to reproduce
call make_lcmv with any invalid weight_norm name e.g. weight_norm='asdf'
Using MNE examples:

```
import matplotlib.pyplot as plt
import numpy as np

import mne
from mne.datasets import sample 
from mne.beamformer import make_lcmv, apply_lcmv

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_raw-eve.fif'
fname_fwd = data_path + '/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif'
label_name = 'Aud-lh'
fname_label = data_path + '/MEG/sample/labels/%s.label' % label_name
subjects_dir = data_path + '/subjects'

event_id, tmin, tmax = 1, -0.2, 0.5

# Setup for reading the raw data
raw = mne.io.read_raw_fif(raw_fname, preload=True)
raw.info['bads'] = ['MEG 2443', 'EEG 053']  # 2 bads channels
events = mne.read_events(event_fname)

# Set up pick list: EEG + MEG - bad channels (modify to your needs)
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=True, eog=True,
                       exclude='bads')

# Pick the channels of interest
raw.pick_channels([raw.ch_names[pick] for pick in picks])
# Re-normalize our empty-room projectors, so they are fine after subselection
raw.info.normalize_proj()

# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax,
                    baseline=(None, 0), preload=True, proj=True,
                    reject=dict(grad=4000e-13, mag=4e-12, eog=150e-6))
evoked = epochs.average()

forward = mne.read_forward_solution(fname_fwd)
forward = mne.convert_forward_solution(forward, surf_ori=True)

# Compute regularized noise and data covariances
noise_cov = mne.compute_covariance(epochs, tmin=tmin, tmax=0, method='shrunk',
                                   rank=None)
data_cov = mne.compute_covariance(epochs, tmin=0.04, tmax=0.15,
                                  method='shrunk', rank=None)

pick_oris = [None, 'normal', 'max-power']
names = ['free', 'normal', 'max-power']
descriptions = ['Free orientation, voxel: %i', 'Normal orientation, voxel: %i',
                'Max-power orientation, voxel: %i']
colors = ['b', 'k', 'r']

fig, ax = plt.subplots(1)
max_voxs = list()
for pick_ori, name, desc, color in zip(pick_oris, names, descriptions, colors):
    # compute unit-noise-gain beamformer with whitening of the leadfield and
    # data (enabled by passing a noise covariance matrix)
    filters = make_lcmv(evoked.info, forward, data_cov, reg=0.05,
                        noise_cov=noise_cov, pick_ori=pick_ori,
                        weight_norm='asdf', rank=None)
    print(filters)
    # apply this spatial filter to source-reconstruct the evoked data
    stc = apply_lcmv(evoked, filters, max_ori_out='signed')

    # View activation time-series in maximum voxel at 100 ms:
    time_idx = stc.time_as_index(0.1)
    max_idx = np.argmax(np.abs(stc.data[:, time_idx]))
    

    if max_idx > len(stc.vertices[0]):
        max_voxs.append(stc.vertices[1][max_idx-len(stc.vertices[0])])
    else:
        max_voxs.append(stc.vertices[0][max_idx])
    
    ax.plot(stc.times, stc.data[max_idx, :], color, label=desc % max_idx)

ax.set(xlabel='Time (ms)', ylabel='LCMV value',
       title='LCMV in maximum voxel')
ax.legend(loc='lower right')
mne.viz.utils.plt_show()
```

#### Expected results
ValueError thrown due to not supported parameter

#### Actual results
There is no error or any unexpected output text if pick_ori is None or 'normal'. Just bad results. If pick_ori == 'max-power' the code crashes on power not being computed due to not passing the if statement in line 387 of _compute_beamformer.py

massich, 2019-02-15:
It does break, the only thing is that the error is not really helpful:

```
UnboundLocalError: local variable 'power' referenced before assignment
```

massich, 2019-02-15:
in the `mne.beamformer.make_lcmv` doc [here](https://martinos.org/mne/dev/generated/mne.beamformer.make_lcmv.html?highlight=weight_norm) it states that `weight_norm` can only be one of these 3 values:

- string `'unit-noise-gain'`
- string `'nai'`
- `None`

agramfort, 2019-02-15:
are you working on this @massich ?

massich, 2019-02-15:
I'm making a PR with the paremeter check, we can discuss further there.

PR #5946 (2019-03-01): FIX: check weight_norm parameter
#### Reference issue
Example: Fixes #5945

#### What does this implement/fix?
checks that `weight_norm : 'unit-noise-gain' | 'nai' | None`

#### Additional information
@jaaval now this snipped errors with a meaningful error

``` python
import mne
from mne.datasets import sample
from mne.beamformer import make_lcmv

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_raw-eve.fif'
fname_fwd = data_path + '/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif'
label_name = 'Aud-l [... cut]
closed 2019-03-01 by wmvanvliet