mne-tools/mne-python #5948: Download status bar shows incorrect speed

cbrnr, 2019-02-16:
I just downloaded the testing data with

```
from mne.datasets import testing
testing.data_path()
```

The download is extremely slow. However, the status bar shown high speed such as

```
[........................................] 100.00% ( 54.9 MB,   7.0 MB/s) -   
```

Obviously, this is incorrect - maybe because the total file size is unknown? Should we do something to fix this?

larsoner, 2019-02-16:
Yeah we should. AFAIK it really only happens with `testing` because it fails to get the size properly sometimes for that dataset in particular, because it comes from GitHub which does some redirects.

cbrnr, 2019-02-16:
OK. Another thing that bothers me is that the status bar doesn't update itself in the same line, but each update creates a new line. Would it make sense to just use `tqdm` for any progress bars we need?

larsoner, 2019-02-16:
It should update on the same line. You are probably looking at a terminal that does not support backspace, so tqdm would not help

cbrnr, 2019-02-16:
You're right, obviously this doesn't work in the PyCharm terminal.

agramfort, 2019-02-16:
I don't know how we can fix this without changing the way we work with the
testing data

>

PR #6684 (2019-08-21): MRG, FIX: More accurate and stable progress bar
Much smoother `MB/s` updates (by scaling the exponential filter coefficent to target a 1 sec time constant), speed verified by monitoring network activity with gnome system monitor. Now will say `unknown` instead of `100.0%` when the file size is wacky.

You can test locally with (which will re-download 1.5GB):
```
python -uic "import mne; mne.datasets.sample.data_path(force_update=True, verbose=True)"
```

Closes #5948.
closed 2019-08-21 by agramfort