mne-tools/mne-python #5951: Bug: Autoreject missing six.py in 18.0

marsipu, 2019-02-16:
#### Describe the bug
When I try to import autoreject in the developement-version of mne-python, an ImportError occurs. I previously have installed autoreject via pip.
The six.py-file seems to be missing in .../mne/externals/

#### Steps and/or code to reproduce
Activate environment of mne-developement-version
pip install autoreject (if not already done)
**Python**
```
>>> import autoreject
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda3\envs\mne_Dev\lib\site-packages\autoreject\__init__.py", line 3, in <module>
    from .autoreject import _GlobalAutoReject, _AutoReject, AutoReject
  File "C:\Anaconda3\envs\mne_Dev\lib\site-packages\autoreject\autoreject.py", line 22, in <module>
    from .utils import (clean_by_interp, interpolate_bads, _get_epochs_type, _pbar,
  File "C:\Anaconda3\envs\mne_Dev\lib\site-packages\autoreject\utils.py", line 15, in <module>
    from mne.externals import six
ImportError: cannot import name 'six'
```
#### Expected results
No errors when importing autoreject (as in 0.17.0)

#### Actual results
An ImportError, six.py is missing in /mne/externals

#### Additional information
Platform:      Windows-10-10.0.17763-SP0
Python:        3.6.8 |Anaconda, Inc.| (default, Feb 11 2019, 15:03:47) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Anaconda3\envs\mne_Dev\python.exe
CPU:           Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 8 cores
Memory:        15.9 GB

mne:           0.18.dev0
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.0
matplotlib:    3.0.2 {backend=Qt5Agg}

sklearn:       0.20.2
nibabel:       2.3.3
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.24.1
dipy:          0.13.0

larsoner, 2019-02-16:
This is a bug with https://github.com/autoreject/autoreject, can you open an issue there?

closed 2019-02-16 by larsoner

marsipu, 2019-02-16:
Oh, I see, sorry. It is already fixed on autoreject by agramfort