mne-tools/mne-python #5952: Make _plot_image public

jona-sassenhagen, 2019-02-17:
evoked and tfr image plots have some fairly elaborate code to plot images with mask and contour. We could expose the (shared) underlying code publicly, greatly simplifying plotting of e.g. statistically thresholded GAT matrices.

larsoner, 2019-02-18:
Okay with me

agramfort, 2019-02-18:
what would be the signature of the public funciton?

jona-sassenhagen, 2019-02-19:
The private function is here:

https://github.com/mne-tools/mne-python/blob/master/mne/viz/utils.py#L2556

Current call signature:

```
def _plot_masked_image(ax, data, times, mask=None, picks=None, yvals=None,
                       cmap="RdBu_r", vmin=None, vmax=None, ylim=None,
                       mask_style="both", mask_alpha=.25, mask_cmap="Greys",
                       yscale="linear"):
```

Probably ax should become a default kwarg ... data is mandatory ... I'd switch times and yvals for xvals and yvals and make both default to None ... maybe add a title, x/ylabel, cbar. So:

```
def _plot_masked_image(data, ax=None, xvals=None, yvals=None, mask=None, picks=None, 
                       cmap="RdBu_r", vmin=None, vmax=None, ylim=None,
                       mask_style="both", mask_alpha=.25, mask_cmap="Greys",
                       yscale="linear", title=None, xlabel=None, ylabel=None, cbar=False):
```

@cbrnr @mmagnuski

mmagnuski, 2019-02-19:
I'm okay with that. Would it be better to call it `plot_image`, not `plot_masked_image` as the mask is optional? Or maybe `plot_image_with_optional_mask_if_you_want_it` ;)

agramfort, 2019-02-19:
ok but please demo this in an existing example

jona-sassenhagen, 2019-02-28:
I will! Thanks guys.