mne-tools/mne-python #5953: simulate_sparse_stc crashes when n_dipoles > len(labels)

QB3, 2019-02-18:
#### Describe the bug
 ```simulate_sparse_stc ```  throws an assertion error when   ```n_dipoles > len(labels) ```.
It should generate   ```min(n_dipoles, len(labels))  ``` dipoles.

#### Steps and/or code to reproduce
If you slightly modified the code of the following MNE example: 
https://www.martinos.org/mne/stable/auto_examples/simulation/plot_simulate_evoked_data.html#sphx-glr-auto-examples-simulation-plot-simulate-evoked-data-py
by remplacing ```n_dipoles=2``` by ```n_dipoles=4``` ligne 58, whiwh I did here:
https://gist.github.com/QB3/6ac76962c9ab7b9f1b44c1026021cafe
an assertion error is thrown.

#### Expected results
As  ```len(labels) = 3  ``` I expect 3 dipoles.

#### Actual results
assertion error from this line https://github.com/mne-tools/mne-python/blob/8b00fc96ec8574c364007ea6b9655eb288d464bb/mne/simulation/source.py#L210

agramfort, 2019-02-18:
I would suggest to raise an informative exception message rather than
silently using min(n_dipoles, len(labels))

>

QB3, 2019-02-18:
OK then we will have to change also the current warning msg:
https://github.com/mne-tools/mne-python/blob/8b00fc96ec8574c364007ea6b9655eb288d464bb/mne/simulation/source.py#L180

agramfort, 2019-02-18:
hum so now you get a warning but it will always crash afterwards? Basically warn should be ValueError or the `min(n_dipoles, len(labels))` should actually be implemented?

QB3, 2019-02-18:
yes, current warning is  ```The number of labels is different from the number of dipoles. 2 dipole(s) will be generated```, but then an assertion error is thrown

agramfort, 2019-02-18:
ok then I would make it a ValueError and not a warn

>

QB3, 2019-02-19:
OK I will do this then:
-  ```n_dipoles > len(labels)```  --->>> ValueError
- ```n_dipoles < len(labels)```  ---->>> keep the same gene

agramfort, 2019-02-19:
go for it @QB3

PR #5976 (2019-02-22): [ENH] ensure simulate_sparse_stc throws error when n_dipoles > len(labels)
 ```simulate_sparse_stc ```  throws an assertion error when   ```n_dipoles > len(labels) ```.
It should generate   ```min(n_dipoles, len(labels))  ``` dipoles.

Fixes #5953
closed 2019-02-22 by massich