mne-tools/mne-python #5961: MNE-C install instructions don't work for macOS 10.14

drammock, 2019-02-19:
The install instructions [here](https://mne-tools.github.io/dev/install_mne_c.html) that worked up through macOS 10.13 apparenty don't work for 10.14. Some brief discussion near end of #5905.

cc @ktavabi @larsoner. Needs testing, @ktavabi are you up for it?  If not, could you at least elaborate here what exactly went wrong?

ktavabi, 2019-02-19:
I didn't really document the errors so I would have to reproduce them. Basically, I just wanted to try and confirm that xquartz gives rise to issues. I hit an error with version 2.7.9 in Rstudio and updated to 2.7.11 which resolved the issue. But during the process, I discovered that mne-c doesn't work with either xquartz versions on my 2018 MBP running OSX 10.14. Right now `mne_analyze` returns _Cannot open display_, with xquartz 2.7.9 it was missing a library file/link IIRC. 
I don't have a lot time to give but I can try and help if need be.

agramfort, 2019-02-19:
maybe @buildqa can help here.

ktavabi, 2019-02-20:
@buildqa I have access to a 2016 MBP with 10.14 on it. I can try MNE-C install if you're unable to. I can do this next week sometime.

buildqa, 2019-02-20:
I've attached the terminal session from following the instructions to install anaconda python 3 (via the command line installer), and then running the mne-python setup instructions.  The vtk python module did not install, and setup.py failed for mayavi.  However, those glitches did not prevent subsequent installs or being able to finish the mne-python setup procedure.   My output from mne.sys_info() looks like it as expected compared to what's listed at the  end of the mne-python  install procedure.  

[mne-python_mojave.txt](https://github.com/mne-tools/mne-python/files/2886952/mne-python_mojave.txt)

larsoner, 2019-02-21:
We are actually curious about MNE-C, not MNE-Python here. In particular I can't find a version of XQuartz that allows the GUIs (e.g. mne_analyze) to work anymore.

buildqa, 2019-02-21:
OK, I should be able to test build the mne-c on 10.14 later today.  I recall from mne-cpp that with Qt used for graphics, the Mac needed to have version 5.10 or higher as older versions of Qt were buggy on the Mac. That does not come on the Mac by default - you can download it from the Qt website or get it thru homebrew.  But  I don't recall now how the older mne-c code base did the GUI's.

larsoner, 2019-02-21:
Yeah, MNE-CPP uses Qt. MNE-C on the other hand uses X calls that on OSX get routed through XQuartz (old tech, no Qt involved at old). From what I recall, on macOS <= 10.13, XQuartz <= 2.7.9 worked, but now on macOS >= 10.14 these old versions of XQuartz no longer work, and MNE-C (mne_analyze, mne_browse_raw) has a display error on newer versions (XQuartz >= 2.7.10). At least that's what I remember from briefly trying to get it to work a while back.

buildqa, 2019-02-22:
I've been building mne-c with cmake and homebrew off a branch - and then off a fork - of Alex's GitHub project (I added some recent changes that Matti made to the fork).  This also allowed for setting up a Mac travis build (which also uses homebrew).  I used the gcc 4.9 compilers via homebrew (and not clang).  I just tried the homebrew/travis build procedure I wrote up on 10.14.0 (with Xquartz 2.7.11) and it works the same as it does on 10.13.6 (with Xquartz 2.7.11).  But when getting that all to work I found that the compiled in lib references to libquicktime, libmat (if you build with the Matlab run time distribution), libgfortran needed to be fixed with install_name_tool, else nothing much would run.  (Running mne_list_versions will find most bad lib references).  libquicktime source is actually in the mne-c source tree and gets built and installed under <install prefix>/lib  I believe it uses X11 to open terminals etc., but I think only mne_analyze and mne_make_movie use it.  If you have access to Alex's repo, you could see what I did, but basically I do an out of source build as described above, so I have a./build subdir parallel to the  ./mne-c source subdir and then set the cmake install prefix to a parallel ./install subdir.   Then from under <install prefix>/bin my install_name_tool commands look like, e.g., for libquicktime, 
$ install_name_tool -change 'libquicktime.dylib' '@loader_path/../lib/libquicktime.dylib' <binary>
(or you could add an rpath instead).

buildqa, 2019-02-22:
I probably should have added that you need the Xquartz 2.7.11 distribution in your path, /opt/X11/bin, which has the X commands

larsoner, 2019-02-22:
> I've been building mne-c with cmake and homebrew off a branch ...

This is useful for developers perhaps, but way out of the reach for users.

Is there currently a solution for using MNE-C for end users on OSX 10.14+? I just downloaded the latest nightly from MGH, and it's build 3378 (the same one I installed at least a few months ago, I think?). When I set `MNE_ROOT` and `source` the `mne_setup_sh` then try `mne_analyze` (with the latest XQuartz, 2.7.11) I get:
```
omar:~ larsoner$ mne_analyze 
Using a 4096-point FFT in filtering of raw data with 2048-sample tapers
Warning: Fatal Error: 
_XmGetDefaultDisplay cannot be used prior to VendorS.Initialize, returns NULL
Segmentation fault: 11
```
Running `mne_list_versions` does indeed show problems:
```
dyld: Library not loaded: libquicktime.dylib
  Referenced from: /Applications/MNE-2.7.4-3378-MacOSX-x86_64/bin/mne_analyze
  Reason: image not found
```
However, `mne_browse_raw` does not suffer from this problem, and I still get the segfault when it tries to open a display:
```
omar:~ larsoner$ mne_browse_raw 
Using a 4096-point FFT in filtering of raw data with 2048-sample tapers
bandpass =    0.0 (   0.0) ...   40.0 (   5.0) Hz
filter is on
64-bit architecture. Using 600.00 Mbytes for ring buffers
Warning: Fatal Error: 
_XmGetDefaultDisplay cannot be used prior to VendorS.Initialize, returns NULL
Segmentation fault: 11
```
When doing either of these things, XQuartz opens just fine, so that distribution at least seems to be working/accessed properly.

But we're digressing a bit, since this issue is really about telling users how to get MNE-C installed and working. Is it correct to as that there is no currently distributed solution for MNE-C on macOS 10.14+? If so, what's the last OSX version currently supported? We should update the docs to say "MNE-C works on OSX <= X and support is in progress for newer versions", and perhaps discuss separately how to actually get it working.

larsoner, 2019-02-22:
Oh! Following the workaround here worked (after Googling the error):

http://spedas.org/wiki/index.php?title=Known_Issues_macOS

In other words:
```
cd /opt/X11/lib
sudo cp libXt.6.dylib libXt.6.dylib.bak
cd flat_namespace/
sudo cp libXt.6.dylib ../.
```
I don't love modifying the XQuartz distribution, but it at least makes things usable. Perhaps we should put this in the docs for now until this can be fixed at the MNE-C build end.

@ktavabi can you confirm it also works for you?

larsoner, 2019-02-22:
(the libquicktime is probably still problematic, but it at least does not stop surface display and probably coreg, raw browsing, etc.)

buildqa, 2019-02-22:
I recall being told the mne-c source code repo using cmake and homebrew should remain private, but I will check about posting the 10.14 binaries.

buildqa, 2019-02-22:
Can you point me to where you got build 3378 from ?

agramfort, 2019-02-22:
even if the repo stays private we should have up to date shipped
binaries for mac an linux.
I've seen many time issues with mismatched versions of libgfortran on
some recent systems for example.
That's why I am pushing for public repos that can leverage CIs for free.
Also if we're for maintaining this code more we should address the
static analysis errors reported.

larsoner, 2019-02-22:
It's the latest build if you log in to the MNE-C download page

http://www.nmr.mgh.harvard.edu/martinos/userInfo/data/MNE_register/index.php

buildqa, 2019-02-25:
I have put in a request to get permission to update the binaries you see when you login to that page.

agramfort, 2019-02-25:
binaries get updated by a cron everynight. Do you plan to change the build
farm locally to fix the cron based packaging?

>

buildqa, 2019-02-25:
The latest  builds I see are: build  3514 for linux from 4/26/2016; build 3420 for MacOS from  10/6/2013.  The Mac build prior to that is 3378 from 6/5/2012 which was mentioned above.  But I don't have permissions to touch/run anything there and I can't read the file that controls what users see after they login.   I was thinking we could use git and travis for any further builds including posting releases of binaries back on GitHub (as a compressed archive).   But I would need accounts with monthly recurring payments on GitHub and Travis in order to keep the source private.  I'm not sure the source code on git can remain private while the binary releases are public.  But if that all worked, maybe we would not need the web page/login mechanism.

agramfort, 2019-02-26:
> The latest builds I see are: build 3514 for linux from 4/26/2016; build 3420 for MacOS from 10/6/2013. The Mac build prior to that is 3378 from 6/5/2012 which was mentioned above.

hum :(

> But I don't have permissions to touch/run anything there and I can't read the file that controls what users see after they login. I was thinking we could use git and travis for any further builds including posting releases of binaries back on GitHub (as a compressed archive).

yes agreed.

> But I would need accounts with monthly recurring payments on GitHub and Travis in order to keep the source private.

yes or we go open source....

> I'm not sure the source code on git can remain private while the binary releases are public. But if that all worked, maybe we would not need the web page/login mechanism.

see with @mshamalainen

drammock, 2019-02-28:
@ktavabi have you had a chance to test the fix that @larsoner mentioned in https://github.com/mne-tools/mne-python/issues/5961#issuecomment-466272857 ?  If it works for you, I'll update the docs accordingly.

ktavabi, 2019-02-28:
@larsoner Sorry this never came up on my radar
Should I be trying this with XQuartz 2.7.11?

> Oh! Following the workaround here worked (after Googling the error):
>
> http://spedas.org/wiki/index.php?title=Known_Issues_macOS
>
> In other words:
>
> ```
> cd /opt/X11/lib
> sudo cp libXt.6.dylib [...]

larsoner, 2019-03-01:
Basically yes, try to get `mne_analyze` or `mne_browse_raw` to run on macOS 10.14+ however it's possible. I think you'll need XQuartz 2.7.11 (since 2.7.9, which used to work on earlier macOS, does not work on 10.14+). But vanilla XQuartz 2.7.11 didn't work for me, I needed the fix linked above.

buildqa, 2019-03-01:
I have uploaded 2 archives to the build area on the server.  Maybe someone could login and let me know if you see the following (and if so download and give it a try).  You will need homebrew installed in the default location,
mne-c_20190301_MacOS-mojave.tgz - build via homebrew on 10.14.0 via gcc 4.9, Xquartz 2.7.11
mne-c_20190301_MacOS-highsierra.tgz - build via homebrew on 10.13.6 via gcc 4.9, Xquartz 2.7.11
The source has some changes from Matti.

Here is what mne_analyze linked against,
$ otool -L mne_analyze
mne_analyze:
    /System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
    /opt/X11/lib/libGL.1.dylib (compatibility version 4.0.0, current version 4.0.0)
    /opt/X11/lib/libX11.6.dylib (compatibility version 10.0.0, current version 10.0.0)
    /usr/local/opt/openmotif/lib/libXm.4.dylib (compatibility version 5.0.0, current version 5.4.0)
    /System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 934.0.0)
    /System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 50.1.0)
    /usr/lib/libexpat.1.dylib (compatibility version 7.0.0, current version 8.0.0)
    /usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
    /opt/X11/lib/libXmu.6.dylib (compatibility version 9.0.0, current version 9.0.0)
    /opt/X11/lib/libXp.6.dylib (compatibility version 9.0.0, current version 9.0.0)
    /opt/X11/lib/libXt.6.dylib (compatibility version 7.0.0, current version 7.0.0)
    /opt/X11/lib/libSM.6.dylib (compatibility version 7.0.0, current version 7.1.0)
    /opt/X11/lib/libICE.6.dylib (compatibility version 10.0.0, current version 10.0.0)
    /opt/X11/lib/libXft.2.dylib (compatibility version 6.0.0, current version 6.2.0)
    /opt/X11/lib/libXext.6.dylib (compatibility version 11.0.0, current version 11.0.0)
    /opt/X11/lib/libpng16.16.dylib (compatibility version 43.0.0, current version 43.0.0)
    /usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
    @loader_path/../lib/libquicktime.dylib (compatibility version 0.0.0, current version 0.0.0)
    /usr/local/opt/libtiff/lib/libtiff.5.dylib (compatibility version 10.0.0, current version 10.0.0)
    /usr/local/opt/gcc@4.9/lib/gcc/4.9/libquadmath.0.dylib (compatibility version 1.0.0, current version 1.0.0)
    /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.200.5)

larsoner, 2019-03-01:
@buildqa did you try running `mne_analyze` to verify it would come up for you on 10.14?

buildqa, 2019-03-01:
Yes I did.  There seems to be some state saved w.r.t the path to the eyes file.  If I had built it in a different git sandbox/branch from last time, it seemed to save the state from the pervious path and claimed it could not find it.  But I did see the whole box/GUI come up on 10.14.

ktavabi, 2019-03-01:
@drammock MNE-C apps work with the XQuartz 2.7.11 modifications as per 
[
http://spedas.org/wiki/index.php?title=Known_Issues_macOS](http://spedas.org/wiki/index.php?title=Known_Issues_macOS)

Additionally, I encountered "Cannot open display" error after the XQuartz install/modification, in which case folks should issue `launchctl load -w /Library/LaunchAgents/org.macosforge.xquartz.startx.plist` before restart. 
This is adapted from  [https://www.xquartz.org/FAQs.html](https://www.xquartz.org/FAQs.html) cc @larsoner 

[image]

buildqa, 2019-03-02:
mne_analyze windows on Mojave (Xquartz 2.7.11, homebrew gcc4.9 build via cmake),
[image]

cjayb, 2019-05-05:
Just confirming that the fix above (copy libXt from flat_namespace) worked on
- macOS 10.14.3
- XQuartz 2.7.11
- mne-C OS X-build 3420 (6 Oct 2013)

Let me know if further testing needed

commit 86b8aeb5 (2019-05-06): closes #5961 (#6274)
closed 2019-05-06 by larsoner