mne-tools/mne-python #5965: BUG?: Channel interpolation: exclude ref_meg in CTF data?

jeythekey, 2019-02-20:
Interpolating channels from CTF data with `_interpolate_bads_meg` results in the interpolated channels having very high amplitude slow fluctuations, if gradient compensation/ref_meg channels are present. I think, similarly to #4663, the ref_meg channels have to be excluded for the reconstruction by setting `ref_meg=False` in the function. What makes me hesitating about this, is that they are actually _explicitly_ included in the moment. Maybe that is related to changes discussed in #5292?

Anyway, I already made a tiny PR which attempts to fix that.

Please let me know if more details/examples are needed (i.e. if this bug is not so obvious).

larsoner, 2019-02-20:
I agree that they should probably be excluded.

jeythekey, 2019-04-26:
Fixed in #6242

closed 2019-04-26 by jeythekey