mne-tools/mne-python #5975: broken read_raw_kit when using test.hsp

massich, 2019-02-21:
Adapted from `mne/io/kit/tests/test_kit.py`
```py
import mne
import numpy as np
import os.path as op

from mne import __file__ as mne_init_path
from mne.io import read_raw_kit
from mne.viz import plot_alignment

KIT_PATH = op.join(op.dirname(mne_init_path), 'io', 'kit', 'tests', 'data')
KIT_HSP = op.join(KIT_PATH, 'test.hsp')
KIT_ELP = op.join(KIT_PATH, 'test.elp')
KIT_SQD = op.join(KIT_PATH, 'test.sqd')
KIT_MRK = op.join(KIT_PATH, 'test_mrk.sqd')

data_path = mne.datasets.sample.data_path()
subjects_dir = data_path + '/subjects'

def get_trans():
    trans = mne.Transform(fro='head', to='mri')
    trans['trans'] = np.array([[0.99981296, -0.00503971,  0.01867181,  0.00255929],
                               [0.00692004,  0.99475515, -0.10205064, -0.02091804],
                               [0.01805957,  0.10216076,  0.99460393, -0.04416016],
                               [0.,          0.,          0.,          1.        ]])
    return trans

raw = read_raw_kit(input_fname=KIT_SQD, mrk=KIT_MRK, elp=KIT_ELP, hsp=KIT_HSP)
fig = plot_alignment(raw.info, trans=get_trans(), subject='fsaverage', dig=True,
                     eeg=['original'], meg=[],
                     coord_frame='head', subjects_dir=subjects_dir,
                     fig=None)

```
[image]

massich, 2019-02-21:
I get the same when using `raw_txt = read_raw_kit(sqd_path, mrk_path, elp_txt_path, hsp_txt_path)`

massich, 2019-02-21:
using `read_dig_montage` is also corrupted.
```py
from mne.channels.montage import read_dig_montage

names = ['nasion', 'lpa', 'rpa', '1', '2', '3', '4', '5']
montage = read_dig_montage(hsp=KIT_HSP,
                           hpi=KIT_MRK,
                           elp=KIT_ELP,
                           point_names=names, transform=False)

my_info = mne.create_info(ch_names=[],
                          sfreq=1,
                          ch_types='eeg',
                          montage=montage)

fig = plot_alignment(my_info, trans=get_trans(), subject='fsaverage', dig=True,
                     eeg=['original'], meg=[],
                     coord_frame='head', subjects_dir=subjects_dir,
                     fig=None)
```

massich, 2019-02-21:
Ok, I think I get it. Both `test.hsp` and `test_hsp.txt` have 500 points, but `test_hsp.txt`'s header says:

```
% 346306 3D points, x y z per line
```
Therefore the files are incomplete, and what the plot shows is a subset of the digpoints (all in the lateral area)

massich, 2019-02-21:
@teonbrooks do you have the complete file? 
@mne-tools/contributors do anyone have a complete hsp file to share (if possible with not lots of points)?

teonbrooks, 2019-02-21:
followed up offline with some files to help

massich, 2019-02-22:
Closing, I'll reopen-and-close if I do a PR updating the test files.

closed 2019-02-22 by massich