mne-tools/mne-python #5977: raw.get_data() should accept int64

skjerns, 2019-02-21:
When loading a file to `RawArray`, the `get_data()` command does only accept `int32`, but should accept all `ints`, including `int64`, as this is the standard `dtype` for many numpy functions (e.g. `np.argmax`). It's a small change, I'll propose it.

agramfort, 2019-02-21:
can you provide a code snippet to replicate? are you on windows?

>

skjerns, 2019-02-21:
yes! sorry, should have done that initially:

```Python
import numpy as np
import mne
from mne.datasets import sample

data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
ch_idx= np.argmax([0,1,2,3,4])
fif = mne.io.read_raw_fif(fname)
data = fif.get_data(ch_idx)
```

only happens with 0.18 dev version! 0.17 seems to be fine.

btw 1.5GB for the sample data set is quite huge!

PR #5978 (2019-02-21): [pick.py] changed checking for int to include int64
#### Reference issue
Fixes #5977 

#### What does this implement/fix?
Many numpy functions return `int64`.
The picks-routine should accept all kind of ints

instead of
```Python
array.dtype!=np.int
```
I implemented
```Python
array.dtype.kind not in ['i', 'u']
```
which checks for any kind of integer
closed 2019-02-21 by larsoner