mne-tools/mne-python #5980: Excessive verbosity during covariance calculation

TwinNeuro, 2019-02-21:
I recently upgraded to Python 3.6 and create an mne environment exactly as per the MNE instructions using the supplied yaml file.

When I use mne.decoding.CSP I get repeated INFO messages:
```
Estimating covariance using EMPIRICAL
Done.
Estimating covariance using EMPIRICAL
Done.
```

#### Describe the bug
When using `mne.decoding.CSP()`  calls so to `mne.compute_covariance()` generates repeating INFO messages that cannot be suppressed by changing logging levels.  Attempts to pass parameters through `cov_method_params={'verbose':None}` also fail.

#### Steps and/or code to reproduce
Run any of the example notebooks on MNE website using the mne.compute_covariance() method, for example:
https://martinos.org/mne/stable/auto_examples/decoding/plot_decoding_csp_eeg.html#sphx-glr-auto-examples-decoding-plot-decoding-csp-eeg-py
The issue exhibits in the browser window as well as in the downloaded ipnb.

In addition, according to the documentation I would expect passing a `cov_method_params={'verbose':None}` dictionary as a parameter to `mne.decoding.CSP()` would allow the verbosity to be suppressed.  Not only does it not, but it returns a very strange error: 
`csp = CSP(n_components=n_components, norm_trace=False, rank='full', cov_method_params={'verbose':False})`
returns:
`ValueError: key (verbose) must be "empirical" or "diagonal_fixed" or "ledoit_wolf" or "oas" or "shrinkage" or "shrunk" or "pca" or "factor_analysis"`

and 
csp = CSP(n_components=n_components, norm_trace=False, rank='full', cov_method_params={'method':'empirical', 'verbose':False})`
returns:
`key (method) must be "empirical" or "diagonal_fixed" or "ledoit_wolf" or "oas" or "shrinkage" or "shrunk" or "pca" or "factor_analysis"`

The issue exhibits in the browser window.

#### Expected results
I expect that the logging should be suppressed if the logging level is set to "WARNING" or "ERROR"

#### Actual results
```
Estimating covariance using EMPIRICAL
Done.
Estimating covariance using EMPIRICAL
Done.
```
or....
```
Estimating covariance using LEDOIT_WOLF
Done.
Estimating covariance using LEDOIT_WOLF
Done.
Estimating covariance using LEDOIT_WOLF
Done.
Estimating covariance using LEDOIT_WOLF
Done.
Estimating covariance using LEDOIT_WOLF
Done.
```
#### Additional information
 `mne.sys_info()`

```
Platform:      Darwin-18.2.0-x86_64-i386-64bit
Python:        3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)  [GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/gerrie/anaconda3/envs/mne/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.17.1
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.2 {backend=nbAgg}

sklearn:       0.20.2
nibabel:       2.3.3
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.24.1
dipy:          0.15.0
```

agramfort, 2019-02-21:
CSP does not expose a verbose parameter. To limit verbosity
use:

mne.set_log_level('warning')

if it does the job please close

>

larsoner, 2019-02-21:
Or open a PR to add the `verbose` parameter to the `CSP` constructor as we do for other classes.

TwinNeuro, 2019-02-27:
Thank you that worked!

closed 2019-02-27 by TwinNeuro