mne-tools/mne-python #5989: Interested to Contrbute to MNE-python

msrdinesh, 2019-02-24:
Hello Everyone, My name is Dinesh. I am third year undergraduate from IIT Bhubaneswar, India. At present I am working on Chaos signal analysis on EEG signals. I am also planning to participate in GSoc'19 with MNE. I have good knowledge in Python,C++ and am specialized in Machine Learning algorithms. I need help to get started with. Please assign me some task/sub goal to work along with you and contribute. Thankyou!

larsoner, 2019-02-24:
Can you take a look at issues marked easy?

msrdinesh, 2019-02-24:
Okay, I will look into them. Actually, I am new to open source contribution. I would be very happy if you could assign me some specific issue to work on so that I can get started. Thanks!

msrdinesh, 2019-02-24:
I have gone through them. Will start working to solve them. Thanks!

closed 2019-02-24 by msrdinesh