mne-tools/mne-python #5992: Unwanted data rejection when running mne.Epochs

YvonneVisser, 2019-02-25:
Detailed instructions on how to file a bug can be found in our [FAQ](https://martinos.org/mne/stable/faq.html#i-think-i-found-a-bug-what-do-i-do).

If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.
Dear MNE developers,

#### The bug

While running the function mne.Epochs, the function is dropping epochs. I'm not asking it to reject data (as far as I can tell) and I can not seem to get it to stop dropping them. Right now, it's rejecting almost all my data, leaving me with only 2 trials.

#### Code snippet

I'm running the function mne.Epochs using the following input parameters:

epochs = mne.Epochs(raw, events, config.event_id, config.tmin, config.tmax,
                        proj=True, picks=picks, baseline=config.baseline, preload=True,
                        decim=config.decim, reject=None)

Where 
config.event_id = {'2s': 2, '4s': 4, '8s': 8, '16s': 16, 'Inf': 32}
config.tmin = -1
config.tmax = 2
config.baseline = (None, 0)
config.decim = 1

#### Code
You can download the variables _raw_ and _events_ used in this code snippet from here:
https://we.tl/t-YT29hjZKzf

#### Expected results
I'm expecting the function to cut the data into epochs, without rejecting data.

#### Actual results

The function runs fine but gives the following (surprising) output:

31 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
Created an SSP operator (subspace dimension = 1)
1 projection items activated
Loading data for 31 events and 3001 original time points ...
**29 bad epochs dropped**

#### Additional information
Output of `mne.sys_info()`:

Platform:      Windows-10-10.0.14393-SP0
Python:        3.6.5 |Anaconda, Inc.| (default, Mar 29 2018, 13:32:41) [MSC v.1900 64 bit (AMD64)]
Executable:    C:\ProgramData\Anaconda2\envs\mne\pythonw.exe
CPU:           Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 8 cores
Memory:        15.9 GB

mne:           0.16.1
numpy:         1.14.3 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=Qt5Agg}

sklearn:       0.19.1
nibabel:       2.2.1
mayavi:        4.6.0 {qt_api=pyqt5}
pycuda:        Not found
skcuda:        Not found
pandas:        0.23.0

Thanks!

larsoner, 2019-02-25:
`events` must be relative to `raw.first_samp`, but they are not:
```
>>> raw = mne.io.read_raw_fif('raw.fif')
>>> print(raw.first_samp)
193000
>>> events = np.load('events.npy')
>>> print(events[:, 0])
[ 28093  39948  54147  59980  71030  82427  91357  96897 108566 119753
 120678 126530 127002 133214 133867 145796 146251 152970 153602 161786
 162280 170639 171224 177875 178303 184103 184526 191305 191807 199033
 199486]
```
Add `raw.first_samp` to `events` and all should work fine

closed 2019-02-25 by larsoner

mmagnuski, 2019-02-25:
@larsoner That might be worth highlighting somewhere in the docs, wdyt? I'm not sure where - there once was "pitfalls" mne page - but now it might be better to stress this in a few tutorials.

larsoner, 2019-02-25:
@drammock feel free to add this to a point of emphasis in the doc work

YvonneVisser, 2019-02-26:
@larsoner 

Thank you for your suggestion.

Unfortunately, I still get the same output:

31 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
Created an SSP operator (subspace dimension = 1)
1 projection items activated
Loading data for 31 events and 3001 original time points ...
29 bad epochs dropped

My events now contain the first sample number:

print(events)
[[ 16000      0      1]
 [ 28093      0      4]
 [ 39948      0     16]
 [ 54147      0     32]
 [ 59980      0      2]
 [ 71030      0     32]
 [ 82427      0     32]
 [ 91357      0      8]
 [ 96897      0      2]
 [108566      0     16]
 [119753      0     16]
 [120678      0     16]
 [126530      0      4]
 [127002      0      4]
 [133214      0     16]
 [133867      0     16]
 [145796      0     16]
 [146251      0     16]
 [152970      0      8]
 [153602      0      8]
 [161786      0      8]
 [162280      0      8]
 [170639      0     16]
 [171224      0     16]
 [177875      0     16]
 [178303      0     16]
 [184103      0      4]
 [184526      0      4]
 [191305      0     16]
 [191807      0     16]
 [199033      0      4]
 [199486      0      4]]

Is there something I am missing?

larsoner, 2019-02-26:
> My events now contain the first sample number:

They do not, `raw.first_samp` for your FIF file is `193000` so all sample numbers `events[:, 0]` should be at least `193000`

YvonneVisser, 2019-02-26:
Ah, apologies, I misunderstood and thought I was supposed to add the first sample number to the events structure. Adding the value of raw.first_samp to each value in the events variable fixed the problem, thank you very much!

The output still says _0 bad epochs dropped_ and I'm not sure I understand it completely. If you have a little more time, could you explain on a higher level why the mne.epoch function is dropping epochs, and why this depends on the sample number? It's just for my understanding so I can prevent problems like this next time!

agramfort, 2019-02-26:
it will drop events that it thinks are outside of the time range of the
data.

looking at epochs.drop_log should inform you about this.

YvonneVisser, 2019-02-27:
Right, thank you @agramfort! That makes sense.