mne-tools/mne-python #5994: Websites with different documentations

msrdinesh, 2019-02-26:
Hello Everyone, my name is Dinesh I am new to Mne-python. I wanted to contribute to MNE and am trying to solve some Doc issues. I have seen that MNE-Documentation has two websites, when I google search about MNE, I get [this](https://martinos.org/mne/stable/index.html). But, there is another site hosted by github, [this](https://mne-tools.github.io/dev/getting_started.html#) one. Some content is different in both the sites. Please let me know how to proceed in this regards. Thanks!

agramfort, 2019-02-26:
https://mne-tools.github.io/dev/getting_started.html#

is updated by our CI while martinos.org is updated manually once in a while.

please don't use the issue tracker to ask such questions. Use gitter
channel or the regular mailing list.

thanks

msrdinesh, 2019-02-26:
Okay, Sorry for that.. thanks for the reply!!

closed 2019-02-26 by msrdinesh