mne-tools/mne-python #5995: read_raw_brainvision stim channel not properly imported to annotations

alexrockhill, 2019-02-26:
I'm not sure where there is a sample brainvision raw file, but if you read it in:

raw = read_raw_brainvision(f,preload=True)

Then the annotations aren't showing up with the data that was contained by the stim channel before:

raw.annotations only gives New Segment/ (1) and Time 0/ (1) whereas this should contain stim channels.

alexrockhill, 2019-02-26:
I should add that this the annotations are properly imported in 0.17.1.

larsoner, 2019-02-26:
@massich can you look into it?

massich, 2019-03-04:
it is weird that our test suite does not capture this. @alexrockhill can you share the file with me? (or even better do you have an MWE that replicates the problem that we can ship with our testdata?)

alexrockhill, 2019-03-04:
In the MNE-sample-data set, I don't see any .vhdr files. If you point me to one, I can make a MWE. Otherwise, I'll send you a file and a gist shortly.

massich, 2019-03-04:
yes I don't find a `.vhdr` file with annotations in our test suite either. We have the annotations in the `.vmrk`. See:

https://github.com/mne-tools/mne-python/blob/194e2e3dc98dd3b545421d5750eff458263c25e7/mne/io/brainvision/tests/test_brainvision.py#L25-L54

We should definitely fix this. But meanwhile:
- can I see what you executed in master and what did you execute in 0.17.1
- can you try to run `read_annotations(vhdr_file)`

alexrockhill, 2019-03-04:
<script src="https://gist.github.com/alexrockhill/9a0e85a38956740ff63d70677d2ca6ad.js"></script>

alexrockhill, 2019-03-04:
In both versions, I just read in the raw file with read_raw_brainvision and then printed to output raw.annotations, I'll check on the read annotations.

massich, 2019-03-04:
This passes in master

```py
    sfreq = 1000.0
    raw = read_raw_brainvision(vhdr_path, preload=True)
    assert raw.annotations.orig_time == 1384359243.794231
    expected = np.array([0, 486., 496., 1769., 1779., 3252., 3262., 4935.,
                         4945., 5999., 6619., 6629., 7629., 7699.]) / sfreq
    description = ['New Segment/',
                   'Stimulus/S253', 'Stimulus/S255', 'Stimulus/S254',
                   'Stimulus/S255', 'Stimulus/S254', 'Stimulus/S255',
                   'Stimulus/S253', 'Stimulus/S255', 'Response/R255',
                   'Stimulus/S254', 'Stimulus/S255',
                   'SyncStatus/Sync On', 'Optic/O  1']
    assert_array_almost_equal(raw.annotations.onset,
                              expected, decimal=7)
    assert_array_equal(raw.annotations.description, description)

```

cbrnr, 2019-03-04:
@alexrockhill you mentioned that this works in 0.17.1 - do you get the issue with the current master?

alexrockhill, 2019-03-04:
I apologize, I didn't record my versions earlier and now with mne 0.18.dev0 I am not able to replicate the same issue, the event markers are coming out fine. I'm not sure why that happened before. If I figure it out I will reopen, sorry!

closed 2019-03-04 by alexrockhill