mne-tools/mne-python #5996: Annotations are not passed on by info in RawArray

alexrockhill, 2019-02-26:
RawArray takes in data, info and an optional first sample. Without reassigning the annotations data field (as shown below), there is no way to pass annotations into the created raw object.

raw = Raw(fname,preload=True)
raw2 = RawArray(raw._data, raw.info.copy())
raw2.annotations = raw.annotations

I think a better solution would be:

raw = Raw(fname,preload=True)
raw2 = RawArray(raw._data, raw.info.copy(),annotations=raw.annotations.copy())

agramfort, 2019-02-27:
raw = RawArray(raw._data, raw.info.copy()).set_annotations(annotations)

is not enough for you?

alexrockhill, 2019-02-27:
I brought this up because there was an incompatibility with the previous version since stim channels were previously in the _data and now they are stored in annotations and thus they were not passed on to the RawArray. It would make sense to me to require all the information to make a raw object using a RawArray especially from a user-friendly perspective of not needing to go through and add data fields (even one) by hand. I thought it might be helpful to bring this up for others who need to debug if existing code uses RawArray and stimulus channels.

closed 2019-02-27 by alexrockhill