mne-tools/mne-python #5997: Standard 10_20 Montage for 22 EEG and 3 EOG channels

Ali-pak, 2019-02-27:
Hello Dear MNE developers, 

I am new to MNE and I am stuck in montages. I have data in .csv and the picture of the montage. Now I don't know how to actually create the montage file, all those angles and distances are missing. Can you guys please suggest me what should I do ? is there a ready to use montage for 22 EEG and 3 EOG headset. ? 
I have created a raw object by using raw.create_info but now When I plot it, it gives all the traces in black and I think the reason is locations of electrodes are missing. 

Thanks in advance

cbrnr, 2019-02-27:
You can get a list of available montages with `mne.channels.get_builtin_montages()` - check if there is a montage which contains your channel names.

Ali-pak, 2019-02-27:
@cbrnr thanks for your comment but none of them have 25 channels, also I have problem adding them to the raw object. The error is the names in Montage are different than the names I am giving as argument.  How can I display the content of Montage ?

Thanks in advance.

Ali-pak, 2019-02-28:
@cbrnr  thanks you are right. It helped now. My montage follow 10-20 :) 
Thanks for your help.

Ali-pak, 2019-03-04:
Hi, @cbrnr I found out that the electrodes position wary with the number of electrodes you select. For example, in my other dataset, I have only 3 electrodes and they should be in the middle but now they are around or even outside the head boundary. do you have any solution for that? 
[image]

agramfort, 2019-03-04:
see https://github.com/mne-tools/mne-python/pull/5472 that is still WIP

>

agramfort, 2019-12-28:
this has now been fixed.

closed 2019-12-28 by agramfort

parasmemon, 2020-06-08:
@cbrnr 
How to rename my channels name according to the "standard_1020" montage?

montage = mne.channels.read_montage("standard_1020")
montage.ch_names

cbrnr, 2020-06-08:
You can rename channels with `raw.rename_channels` (https://mne.tools/stable/generated/mne.rename_channels.html) and provide either a dict or a callable to rename your existing channels names.

parasmemon, 2020-06-08:
@cbrnr 
Description of my data: "According to the standard 10-20 EEG location system, on both right and left hemispheres, eight four-millimeter electrodes were placed in an array along important language production (Broca’s area; F3 and F4) and language comprehension (Wernicke’s area; P5 and P6) regions of the human brain 9 (N=32)."
Montage Channels Names: 
['LPA', 'RPA', 'Nz', 'Fp1', 'Fpz', 'Fp2', 'AF9', 'AF7', 'AF5', 'AF3', 'AF1', 'AFz', 'AF2', 'AF4', 'AF6', 'AF8', 'AF10', 'F9', 'F7', 'F5', 'F3', 'F1', 'Fz', 'F2', 'F4', 'F6', 'F8', 'F10', 'FT9', 'FT7', 'FC5', 'FC3', 'FC1', 'FCz', 'FC2', 'FC4', 'FC6', 'FT8', 'FT10', 'T9', 'T7', 'C5', 'C3', 'C1', 'Cz', 'C2', 'C4', 'C6', 'T8', 'T10', 'TP9', 'TP7', 'CP5', 'CP3', 'CP1', 'CPz', 'CP2', 'CP4', 'CP6', 'TP8', 'TP10', 'P9', 'P7', 'P5', 'P3', 'P1', 'Pz', 'P2', 'P4', 'P6', 'P8', 'P10', 'PO9', 'PO7', 'PO5', 'PO3', 'PO1', 'POz', 'PO2', 'PO4', 'PO6', 'PO8', 'PO10', 'O1', 'Oz', 'O2', 'O9', 'Iz', 'O10', 'T3', 'T5', 'T4', 'T6', 'M1', 'M2', 'A1', 'A2']... These are 97 channels.
My data Channels Names: 
In my data channels names are [1,2,3,...,40]
If I am starting with renaming one channel name:

Renaming Code: mne.rename_channels(raw, ('1', 'LPA'))

Error: AttributeError: 'RawBrainVision' object has no attribute '_check_consistency'

Q#01: I am getting this error?
Q#02: How do I know which channel name is replaced with the montage channel name?

drammock, 2020-06-08:
> Renaming Code: mne.rename_channels(raw, ('1', 'LPA'))

@parasmemon if you look more closely at the docstring of `rename_channels`, you'll see that it expects an **info object** and a **dictionary**.  Your code sample (quoted above) is passing a **raw object** and a **tuple**.

> How do I know which channel name is replaced with the montage channel name?

You will need to figure that out yourself by looking at the documentation for your EEG system/cap, and possibly compare it to an image of the standard 10-20 system.

Please don't post any more questions to this thread; this issue has been closed for several months and is not the right place to ask usage questions. For further help, please use the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python) if you need additional help.

parasmemon, 2020-06-08:
@drammock 

I tried this:
mne.rename_channels(raw.info, {'1': 'LPA'})

Again getting an error:
ValueError: Channel name(s) in mapping missing from info: ['1']

drammock, 2020-06-08:
@parasmemon repeating myself:

> Please don't post any more questions to this thread; this issue has been closed for several months and is not the right place to ask usage questions. For further help, please use the [MNE mailing [...]