mne-tools/mne-python #6001: Add option to save/export raw to .mat

cbrnr, 2019-02-28:
I'm thinking of adding an option to save a raw object (or at least the most essential information, such as the data, channel names, sampling frequency, etc.) to a .mat file in [MNELAB](https://github.com/cbrnr/mnelab). However, this might be something that could also be added directly to MNE. I'd probably add a new `raw.export` method, because `raw.save` saves all information to a FIFF file whereas I'm planning to export just the most essential fields (at least for now).

WDYT? Would this be something for MNE or should I rather add it to MNELAB?

larsoner, 2019-02-28:
Someday we might want to support writing to other formats, too (e.g., EDF), so we'd have to think about the API we'd want to use. I'd be fine with just triaging based on file extension. But ideally whatever we wrote would write out all of the `meas_info` / be as complete as FIF. Maybe the thing to do is have it in MNELAB for now to see how it works, and once it's mature port it over?

agramfort, 2019-02-28:
you mean write proper EEGLAB or fieldtrip files?

cbrnr, 2019-03-01:
@agramfort I was thinking of writing EEGLAB files, because I don't want to invent yet another .mat-based format. This means that we'd populate as many fields as we can, but in general information from one format can't be mapped 1:1 onto another format, so we probably can't write out all of `meas_info`. But you're right, maybe the best approach is to include it in MNELAB for now because I will be writing out the minimum number of fields necessary for a valid EEGLAB file.

agramfort, 2019-03-01:
agreed

>

cbrnr, 2019-03-01:
I've implemented it in MNELAB. Basic export to .set works, including the data, channel names, sampling frequency, and events (converted from annotations). If anyone wants to give it a try and let me know if it works I'd be very happy. Note that I expect that this works best with MNE 0.18, because otherwise synthesized stim channels get in the way. If we want that in MNE, I'm happy to port it over.

cbrnr, 2019-03-12:
FYI, export to EDF also works in MNELAB (if `pyedflib` is installed), including annotations. I just didn't find a way to write non-integer duration files (e.g. 20.2 seconds), because `pyedflib` always seems to write in increments of 1 second...

closed 2019-03-12 by cbrnr