mne-tools/mne-python #6009: Open a FIF file using in-memory binary streams

paulroujansky, 2019-03-01:
Hello,

I am struggling with this one : I have stored some raw EEG data (in ".fif" format) on a remote host which I would like to load through a SFTP connection, without copying the data on the disk of my local machine (in order to save space/gain computation time).
In order to perform this, I am using the library `paramiko`, in order to copy the data in an in-memory bytes buffer (a `io.BytesIO` instance).
My question is : is there a way to directly load the content of the buffer into a `Raw` object using `mne.io.read_raw_fif`?

The code is roughly the following :
```python
with io.BytesIO() as fl:
    # Copy remote file (given by its remotepath) from the SFTP server and write it to
    # file-like object `fl`.
    sftp_client._getfo('/my/path/on/remote/host/my_raw_file.fif', fl, **kwargs)
    raw = read_raw_fif(fl, preload=True, verbose=True)
```

I have come to the conclusion that there is not, but I wondered if there was some kind of hack, or if I simply missed something here.

`read_raw_fif` expects a filename as a `str`, but actually `mne.io.Raw` support open-file opening through `fiff_open` (but one cannot pass a `io.BytesIO` object in the first place though...)

Please tell me if you need further details.

agramfort, 2019-03-01:
did you consider rather use sshfs to mount a virtual disk?

indeed what you propose cannot work as it is now.

>

larsoner, 2019-03-01:
We decided not to support this (at least for now) in #5867, feel free to look at that discussion

closed 2019-03-01 by larsoner