mne-tools/mne-python #6027: Artifacts are still there even after  applying ICA

Ali-pak, 2019-03-04:
Hello Dear mne developers, 
I am trying remove artifacts from my data and I have 3 EOG channels but my data after and before the procedure looks the same. I am following the example code with a little change, below is my code. 
mne_analyze listserv

```
raw.filter(1., None, n_jobs=1, fir_design='firwin')
        picks_eeg = mne.pick_types(raw.info, meg=False, eeg=True, eog=True,stim=False,exclude='bads')
        method = 'fastica'
        # Choose other parameters
        n_components = 6 
        decim = 3       
        ica = ICA(n_components=n_components, method=method)
        reject = dict(mag=5e-12, grad=4000e-13)
        ica.fit(raw, picks=picks_eeg, decim=decim, reject=reject)
        eog_inds, scores = ica.find_bads_eog(raw, ch_name=None)
        ica.exclude.extend(eog_inds)
        raw.plot(n_channels=3, scalings='auto', title='Auto-scaled Data from arrays',show=True,  block=True)
        raw_copy = raw.copy()
        ica.apply(raw_copy)
        raw_copy.plot(n_channels=3, scalings='auto', title='Auto-scaled Data from arrays',show=True, block=True)
        raw_copy=raw_copy.get_data(picks=None, start=0, stop=None, reject_by_annotation=None, 
                          return_times=False)
```

Kindly direct me in the right direction.  Thanks :)

cbrnr, 2019-03-05:
Can you post images of segment with eye blinks before and after ICA? Are they exactly the same?

cbrnr, 2019-03-05:
Also, could you show the contents of `eog_inds` as well as scalp maps of these ICs?

jona-sassenhagen, 2019-03-23:
@Ali-pak are you still there?

jona-sassenhagen, 2019-04-27:
Closing. @Ali-pak reopen if you have updates.

closed 2019-04-27 by jona-sassenhagen