mne-tools/mne-python #6030: layout.plot(picks=???)

drammock, 2019-03-05:
I noticed that the `picks` argument to `mne.channels.Layout.plot` doesn't seem to work as expected based on the docs.  First I was working with a layout loaded from file:

```Python
import mne
mne.channels.read_layout('Vectorview-all').plot(picks=['mag'])
# ValueError: picks as str can only be used when measurement info is available
```

OK, so I guess that `raw.info['dig']` is what it needs?  But I get the same result when deriving the layout from `raw.info` of the sample data:

```Python
fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
mne.find_layout(raw.info).plot(picks=['mag'])  # same value error as above
```
Same thing happens if I try to pass specific channel names instead of a channel type.  Is this a bug or am I just being dumb here?

#### Expected results
```Python
mne.channels.read_layout('Vectorview-all').plot(picks=['mag'])
# ^^^ this should look identical to this vvv
mne.channels.read_layout('Vectorview-mag').plot()
```
and this should work
```Python
mne.channels.read_layout('biosemi').plot(picks=['Fz', 'Cz', 'Pz', 'Oz'])
```

#### Actual results
```
In [107]: mne.find_layout(raw.info).plot(picks='mag')                                                                                 
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-107-7fb1eb6c1795> in <module>
----> 1 mne.find_layout(raw.info).plot(picks='mag')

/opt/mne-python/mne/channels/layout.py in plot(self, picks, show)
    110         """
    111         from ..viz.topomap import plot_layout
--> 112         return plot_layout(self, picks=picks, show=show)
    113 
    114 

/opt/mne-python/mne/viz/topomap.py in plot_layout(layout, picks, show)
   2158     pos, outlines = _check_outlines(pos, 'head')
   2159     _draw_outlines(ax, outlines)
-> 2160     picks = _picks_to_idx(len(layout.names), picks)
   2161     pos = pos[picks]
   2162     names = np.array(layout.names)[picks]

/opt/mne-python/mne/io/pick.py in _picks_to_idx(info, picks, none, exclude, allow_empty, with_ref_meg)
    829         raise ValueError('picks must be 1D, got %sD' % (picks.ndim,))
    830     if picks.dtype.char in ('S', 'U'):
--> 831         picks = _picks_str_to_idx(info, picks, exclude, with_ref_meg)
    832     if picks.dtype.kind not in ['i', 'u']:
    833         raise TypeError('picks must be a list of int or list of str, got '

/opt/mne-python/mne/io/pick.py in _picks_str_to_idx(info, picks, exclude, with_ref_meg)
    855     # special case for _picks_to_idx w/no info: shouldn't really happen
    856     if isinstance(info, int):
--> 857         raise ValueError('picks as str can only be used when measurement '
    858                          'info is available')
    859 

ValueError: picks as str can only be used when measurement info is available
```

#### Additional information
```
Platform:      Linux-4.15.0-45-generic-x86_64-with-debian-buster-sid
Python:        3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44)  [GCC 7.3.0]
Executable:    /opt/miniconda3/envs/mnedev/bin/python
CPU:           x86_64: 8 cores
Memory:        7.6 GB

mne:           0.18.dev0
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    3.0.2 {backend=Qt5Agg}

sklearn:       0.20.2
nibabel:       2.3.1
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.23.4
dipy:          0.15.0
```

larsoner, 2019-03-05:
> OK, so I guess that raw.info['dig'] is what it needs? 

No, `info['chs']` to know what is mag, grad, eeg, etc.

> But I get the same result when deriving the layout from raw.info of the sample data:

When you call `find_layout`, the returned `layout` object does not keep the `info` around. We could either make it keep `info`, or (probably better) add `info` as an optional argument to `plot_layout` to work around this to allow `str` arguments for `picks`.

drammock, 2019-03-05:
OK, so is this a case of "let's fix the docs to match the code" or "let's fix the code to match the docs"?  The docs are pretty clear that this should already work:
```
picks : str | list | slice | None
    Channels to include. Slices and lists of integers will be
    interpreted as channel indices. In lists, channel *type* strings
    (e.g., ``['meg', 'eeg']``) will pick channels of those
    types, channel *name* strings (e.g., ``['MEG0111', 'MEG2623']``
    will pick the given channels. Can also be the string values
    "all" to pick all channels, or "data" to pick data channels.
    None (default) will pick all channels.
```

larsoner, 2019-03-05:
We filled in the docstring with the wrong template in this case. It should use the one that does not say anything about `info`, which is `picks_nostr` (I think). That's the easier fix, then if someone wants the `str` support they can add it and change which template var is used.

PR #6187 (2019-04-23): fix doc on plot_layout
closes #6030

as we don't have an Info to pick by types.
closed 2019-04-23 by larsoner