mne-tools/mne-python #6032: 4D/BTi import error if pdf/config channel orders to not match

joshbear, 2019-03-05:
#### Describe the bug
If the order of channels is different in the config and header (pdf) files, than the current match algorithm will report that they do not match even if all channels are present in both. 

#### Steps and/or code to reproduce
You need to have a config and header file with channels in a different order. If you do, then all you need to do is:

raw = mne.io.read_raw_bti(filename) 

#### Expected results
import the file

#### Actual results
throws a runtime exception

#### Additional information
Platform:      Darwin-18.2.0-x86_64-i386-64bit
Python:        3.6.5 |Anaconda, Inc.| (default, Apr 26 2018, 08:42:37)  [GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/joshbear/bin/anaconda3/envs/mne/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.17.0
numpy:         1.14.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=MacOSX}

sklearn:       0.19.1
nibabel:       2.2.1
mayavi:        4.5.1.dev0 {qt_api=pyqt5, PyQt5=5.10.1}
cupy:          Not found
pandas:        0.23.0
dipy:          Not found

agramfort, 2019-03-06:
maybe @pravsripad or @jdammers can help?

jdammers, 2019-03-06:
@joshbear would you mind sharing the files for testing?

joshbear, 2019-03-06:
@jdammers Try this link: https://1drv.ms/u/s!AjNKUsqPwOwNgVQU6qT5fnX4lyXt
It's about 75 mb. 10 minute recording. Let me know if problems.

jdammers, 2019-03-07:
@joshbear I can confirm that the number of channels found is the same in the pdf and the config. However, as you mentioned, the channel order in the pdf is different (kind of unsorted) as compared to the config (channels are sorted). The current implementation of read_raw_bti() compares each entry of the pdf channel list with the channel list in the config file (i.e., a 1-by-1 matching) and therefore it complains about the unmatch.  What is the reason for the different channel order in the pdf?

joshbear, 2019-03-07:
Honestly, I'm not sure. I've tried to figure that out from our physicist, but his initial impression was that it is just a quirk in MNE. However, I've processed a couple dozen files, and this one run from this one subject is the only one that seemed off. Indeed, I've recently (pleasantly) discovered that the other runs from the same recording don't have this problem. It could be a one-off quirk, and I can discard this run (for my current purposes, it won't really matter that much — when I thought it was affecting all runs, it was a bigger problem). but it will raises the question of whether there is some way that the channels could be properly re-ordered without introducing new errors?

jdammers, 2019-03-07:
I will ask my colleague Eberhard Eich, he the real experts. Maybe Eberhard has an idea how to rearrange the channel order in the pdf. We will let you know soon.

jdammers, 2019-03-13:
@joshbear please have a look at #6048

PR #6300 (2019-09-20): MRG, FIX: BTi reading bugs
@joshbear @ebeich can you 1) see if this works on your data and 2) check that you're happy with the name and URL I put in `whats_new.rst` (based on 2 minutes of Google investigation)? I just cherry-picked your commits and updated `whats_new.rst`. It would be nice to have some test data for these use cases but might be overkill for this corner case.

Closes #6048.
Closes #6033.
Closes #6032.
closed 2019-09-20 by larsoner