mne-tools/mne-python #6034: raw.plot() axes overlap on macOS with qt5 backend

cbrnr, 2019-03-06:
When I plot raw data on macOS using the `qt5` Matplotlib backend, some axes overlap and are therefore unreadable (the time axis labels of the currently visible segment, the "Ch." label, the channel labels, the x axis label, and even the "Help" button):

[image]

Everything works when I use the default `osx` backend though:

[image]

Is this something we can fix? Where would you start debugging this in the first place? BTW, everything looks normal on Linux using the `qt5` backend, so it seems to be specific to macOS (I don't have Windows to test this).

larsoner, 2019-03-06:
Probably an issue with HiDPI / physical vs logical pixels. I'd start out by searching terms related to this to find issues like https://github.com/matplotlib/matplotlib/issues/8052.

If nothing promising having to do with HiDPI / scaling / physical vs logical pixels comes up, I'd try to replicate using a minimal example that just uses `plt` calls that mimic what we do in `raw.plot` (which appears to be at least constructing axes using `plt.subplot2grid`). Then once the MWE is set, ping the `matplotlib` listserv to see the right way to deal with it.

agramfort, 2019-03-06:
I confirm the problem on my mac

>

cbrnr, 2019-03-07:
@larsoner you're right, this is definitely related to HiDPI resolutions. I usually scale my 3840x2160 display to 3360x1890, but if I turn off scaling (that is, if I use the full 4K resolution), the plot looks fine.

agramfort, 2019-03-07:
are you doing this on a laptop screen too? on my mac book with default
settings I have the pb.

cbrnr, 2019-03-07:
Yes, the default on MacBooks with retina display is also some scaled resolution.

cbrnr, 2019-03-08:
I couldn't come up with a simple MWE, but I found that maybe we can fix the problem in https://github.com/mne-tools/mne-python/blob/56ce204fda6763458540755988f97c76a69bbcbd/mne/viz/utils.py#L472 ourselves. I tested with 5 resolutions and 2 backends (Qt5Agg and macosx), here are the values for the sizes we compute:

```
Qt5Agg
======

1:1 3840x2160:
size = [1620.  939.], get_size_inches = [16.2   9.39], dpi = 100.0
pos = [0.06172839506172839, 0.10117145899893504, 0.9104938271604939, 0.8615548455804047]

Scaled 3360x1890:
size = [3240. 1878.], get_size_inches = [16.2   9.39], dpi = 200.0
pos = [0.030864197530864196, 0.05058572949946752, 0.9552469135802468, 0.9307774227902024]

Scaled 3008x1692:
size = [3240. 1878.], get_size_inches = [16.2   9.39], dpi = 200.0
pos = [0.030864197530864196, 0.05058572949946752, 0.9552469135802468, 0.9307774227902024]

Scaled 2560x1440:
size = [3240. 1878.], get_size_inches = [16.2   9.39], dpi = 200.0
pos = [0.030864197530864196, 0.05058572949946752, 0.9552469135802468, 0.9307774227902024]

2:1 1920x1080:
size = [3240. 1878.], get_size_inches = [16.2   9.39], dpi = 200.0
pos = [0.030864197530864196, 0.05058572949946752, 0.9552469135802468, 0.9307774227902024]

macosx
======

1:1 3840x2160:
size = [1620.  939.], get_size_inches = [16.2   9.39], dpi = 100.0
[0.06172839506172839, 0.10117145899893504, 0.9104938271604939, 0.8615548455804047]

Scaled 3360x1890:
size = [1620.  939.], get_size_inches = [16.2   9.39], dpi = 100.0
pos = [0.06172839506172839, 0.10117145899893504, 0.9104938271604939, 0.8615548455804047]

Scaled 3008x1692:
size = [1620.  939.], get_size_inches = [16.2   9.39], dpi = 100.0
pos = [0.06172839506172839, 0.10117145899893504, 0.9104938271604939, 0.8615548455804047]

Scaled 2560x1440:
size = [1620.  939.], get_size_inches = [16.2   9.39], dpi = 100.0
[0.06172839506172839, 0.10117145899893504, 0.9104938271604939, 0.8615548455804047]

2:1 1920x1080:
size = [1620.  939.], get_size_inches = [16.2   9.39], dpi = 100.0
pos = [0.06172839506172839, 0.10117145899893504, 0.9104938271604939, 0.8615548455804047]
```

With the `macosx` backend, the figure is always correct. With the `Qt5Agg` backend, the figure displays correctly only using the native 1:1 resolution. All other resolutions yield twice the `dpi` and `size`. I wonder if we can use this somehow in our border calcuations...

larsoner, 2019-03-08:
You can use `dpi_ratio = getattr(fig.canvas, '_dpi_ratio', 1.)` to get the HiDPI ratio. Can you see if there is a way to fix our spacing using this value?

cbrnr, 2019-03-10:
@larsoner can you test #6043 to see if this works for you?

closed 2019-03-10 by larsoner