mne-tools/mne-python #6038: MLP low accuracy and strange confusion matrix in case of multi-classification

michaelmart77, 2019-03-07:
I'm trying the MLP from sklearn.neural_network import MLPClassifier for multi classification of about 15 classes. The classification is done on CSP (common spatial patterns) components from EEG Data. I compared the MLP with Logistic regression from sklearn.linear_model import LogisticRegression
The MLP was far worse than the logistic regression.
this is part of my code for:

from sklearn.neural_network import MLPClassifier
my_ANN = MLPClassifier(hidden_layer_sizes=(100,80,70,50,40,2), activation= 'relu', 
                       solver='adam', alpha=1e-6, max_iter=2000,batch_size=1,
                       learning_rate_init = 0.1)
csp = CSP(n_components=20,reg=None)
clf = Pipeline([('CSP', csp), ('MLP', my_ANN)])
ANN_score = cross_val_score(clf, X_train, y_train, cv=2, n_jobs=1, scoring='accuracy')

Also I'm attaching both confusion matrices in the MLP and Logistic Regression case. In the logistic regression case there are some scattered labels but there are also labels on the diagonal which is good and the accuracy wasn't that good but at least reasonable. In the MLP case the labels are very strange. Not scattererd all true labels are matching two predicted labels in two columns. So now I have the feeling that there's something wrong with MLP. Is the code wrong? Is the implementation is wrong? Is the MLP not suitable for multi-classification? How to solve this in the case of the MLP?

[image]
[image]

mmagnuski, 2019-03-07:
This question seems to be better fit for the mailing list - you might get more responses there.
Did you try using a simpler network? Comparing vanilla logistic regression with a neural network with so many neurons and layers (`hidden_layer_sizes=(100,80,70,50,40,15)`) may not give you a good idea for why the performance is different.

jona-sassenhagen, 2019-04-25:
Can this be closed?

larsoner, 2019-04-25:
Yes let's move to the mailing list

closed 2019-04-25 by larsoner