mne-tools/mne-python #6039: y-limits of subplots in plot_evoked_topo

flecaignard, 2019-03-07:
Hi all,

Is there a way to set y-axis limits in the « click-to-magnify » plots in plot_evoked_topo? The ylim input applies to the main plot, but does not seem to be transferred to sensor-specific subplots.
Thanks!

Françoise

jjnurminen, 2019-03-11:
Thanks for the ping @agramfort 
I can take a look, if this wasn't picked up already.

agramfort, 2019-03-11:
please go for it Jussi

>

jjnurminen, 2019-03-12:
I started playing with this, but can't get the ylim to work at all on latest master. For example, modifying `plot_topo_compare_conditions` and supplying `ylim` as below has no effect on the plot. Am I missing something?

    plot_evoked_topo(evokeds, color=colors, title=title, background_color='w', ylim=dict(grad=[-1e3, 1e3], mag=[-1e3, 1e3]))

jjnurminen, 2019-03-12:
Also it looks like topoplot refactor is coming in #5568. I guess that will completely change how ylim works, so maybe we should wait.
@jona-sassenhagen

jona-sassenhagen, 2019-03-12:
I will try to get the refactor done at the code sprint ... But I am not yet sure how to deal with interactivity.

agramfort, 2019-03-12:
maybe this can be done during the sprint then...

>

jona-sassenhagen, 2019-03-12:
We can try. @flecaignard can you wait for a month? :)

jona-sassenhagen, 2019-04-26:
This is done in  #6211. However,  #6211 is not doing evoked.plot_topo (but a way for doing topo plots with viz.plot_compare_evokeds). Should I additionally work on making evoked.plot_topo handle this?

PR #6211 (2019-04-27): MRG +1 plot_compare_evokeds topo functionality
Another attempt. It's not really hard, but it's my 3rd attempt so ...

mne.viz.plot_compare_evokeds gains the ability to plot to a topographical layout.
This will be slow for MEG data, but is often used for ERP papers.

Not ready for reviews yet.

Closes #4746
Closes #5746
Closes #6166
Closes #6039

Redo of #5568
closed 2019-04-27 by agramfort