mne-tools/mne-python #6054: brainvision _str_to_meas_date breaks with incorrect date

petergrey75, 2019-03-12:
if the original brainvision file lacks date information and we do segmentation on it, the offset of "New Segment" is added to "000000000000000000" so we end with something like '00000000000304125000' which passes sanity check in _str_to_meas_date, but it is stil invalid:

`def _str_to_meas_date(date_str):
    date_str = date_str.strip()

    if date_str in ['0', '00000000000000000000']:
        return None`

the solution is to add additional check:

`    try:
        meas_date = datetime.strptime(date_str, '%Y%m%d%H%M%S%f')
    except ValueError:
        return None`

agramfort, 2019-03-13:
@petergrey75 can you share a full code snipped with data so we can replicate?
it will be used then as a non-regression test in the PR. thanks

petergrey75, 2019-03-13:
@agramfort the only change is in brainvision.py in _str_to_meas_date function.

```
def _str_to_meas_date(date_str):
    date_str = date_str.strip()

    if date_str in ['0', '00000000000000000000']:
        return None
	
    try:
        meas_date = datetime.strptime(date_str, '%Y%m%d%H%M%S%f')
    except ValueError:
        return None
```

I just added try/except clause around strptime function because if the date_str is invalid it breaks the loading of the file.

The problem is that segmentation of the file without (empty) meas_date just adds the offset value of the 1st segment to the initial date_str (which is '00000000000000000000' in that case) then the new date_str is something like '00000000000304125000' which pass the first check but will cause ValueError on strptime.

agramfort, 2019-03-13:
we don't do changes in code without adding a unit test.

can you share a code snippet that would break in current master and that
works with your suggested  change? thanks

>

petergrey75, 2019-03-13:
i can share the code, but unfortunately i'm not allowed to share the data, so code without the data is meaningless, i guess?

larsoner, 2019-03-13:
Within a test, you can copy `shutil.copyfile` an existing `.eeg` test file and then `with open(fname, 'r') as fid:` "copy" the `.vhdr` file with an appropriate modification that would show the error on `master` but is fixed in this PR.

I know it might seem a bit overkill for a simple fix that is easily explained, but the 10-20 lines necessary to make this test will help maintainability for us.

If what I'm suggesting isn't clear or doesn't make sense I can try to push a commit to your branch to do it.

PR #6188 (2019-04-23): MRG: Fix BrainVision date error
Fixes #6054 

Changes:

- Removing a superfluous  conditional check that is already handled shortly afterwards 
- Introducing a test case that reproduces the error described by @petergrey75 in #6054 
- Inserting a try-except clause in `_str_to_meas_date` to fix the error
closed 2019-04-23 by larsoner