mne-tools/mne-python #6062: Do we need scikit-learn for Infomax ICA?

cbrnr, 2019-03-15:
Currently, we [require `scikit-learn` for all ICA algorithms](https://github.com/mne-tools/mne-python/blob/cffcf4b1aeaf04655f6917718029c59ff412c917/mne/preprocessing/ica.py#L316). However, I don't think that Infomax requires `scikit-learn`, only FastICA uses the implementation in this package. Not sure how we handle Picard, but I think it requires `scikit-learn` either.

agramfort, 2019-03-15:
true: infomax and picard don't need sklearn (see
https://github.com/pierreablin/picard/blob/master/.travis.yml)

>

cbrnr, 2019-03-19:
I'm not sure that this was correct. It looks like we need PCA from scikit-learn, no matter what:
https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L613

How should we proceed?

larsoner, 2019-03-19:
We can add our own implementation of PCA pretty easily. IIRC sklearn has a bunch of options I think we never use (sparse, randomized, etc.), and we can get away with just making an efficient one using `scipy.linalg`/SVD. Okay with you @agramfort ?

agramfort, 2019-03-19:
yes ok with me and there is also a long standing issue in there. scikit-learn centers the data all the time while we may not want to do this and I think eeglab never centers the signals that are anyway zero mean after highpass filtering. We saw the problem with @massich and @pierreablin when apply ICA on cropped and short signals.

massich, 2019-03-19:
+1 to add our own PCA. IIRC we only need a subset of sklearn's PCA, if so we should be able to swap those two and nothing break. (and maybe be even numerically stable)

cbrnr, 2019-03-19:
I agree, PCA is one line of code if we use e.g. `np.linalg.svd`, but the real work is making sure this yields the same results like the scikit-learn implementation. Who's up for it (I don't have time ATM but I'm happy to review).

massich, 2019-03-19:
I can put it on my list.

mmagnuski, 2019-03-21:
@agramfort - could you elaborate a little bit on the mean-centering problems? I was looking through the eeglab code recently and they do mean centering at the beginning:
https://github.com/sccn/eeglab/blob/7a93f0276ed6e42fa59cbd49a55f9b5d3f22e707/functions/sigprocfunc/runica.m#L657

massich, 2019-03-21:
@mmagnuski this is exactly it. I'll try to elaborate but if I'm really off please correct me.

If you run PCA and you have DC in your signal, then the first component of the PCA what is mainly doing is to pick up this DC.

The PCA in sklearn IIRC does not do this. It assumes that your data is already centered to give more control to the user.

The typical case, from the training data you learn the translation (the mean) and the axis rotation (PCA) then to predict in new data you translate your data and rotate it ( `apply_PCA(data- mean(training))`).
Imagine a case where your sensors have drift or inertia (i.e. some light captors). The sensor still picks variation but it keeps accumulating some photons that is not able to wash away. For a short interval, the data is perfectly fine but if you have trained with chunks that are really far from your testing, the results would be bad. That's why you can remove the mean of both sets independently and assume that it has always been 0. (or what is the same, apply a high pass filtering)

mmagnuski, 2019-03-21:
@massich 
My confusion stemmed from Alex saying that sklearn always centers and eeglab does not, while I checked that eeglab does center the data. Now you say that sklearn PCA does not center and I'm a bit more confused. :)
But I understand your argumentation - although given we often highpass filter at 1 Hz is it a realistic scenario? Also - you seem to refer to training and testing datasets here, which is not a common practice with ICA - you train and apply the ICA on the same data in most scenarios.

agramfort, 2019-03-22:
indeed @cbrnr eeglab does center the data as well. I think this is a bad idea if you have cropped filtered data as we saw last year with @massich and @pierreablin. I would vote to make it an option (center=True or False and make it True by default)

larsoner, 2019-03-22:
@massich I can do this tomorrow if you haven't started

massich, 2019-03-22:
Go for it

PR #6063 (2019-03-15): Require sklearn only for FastICA
Fixes #6062.
PR #6081 (2019-03-26): MRG: Vendor simplified PCA
Closes #6062.
closed 2019-03-26 by cbrnr