mne-tools/mne-python #6069: Issues computing connectivity with CTF system

jasmainak, 2019-03-19:
I got a report that the connectivity computation was problematic. If you have a CTF-270 system, the function find_ch_connectivity gives you the connectivity of the CTF-275 system. Code reproduce:

```py
import numpy as np

import mne
from mne.channels import find_ch_connectivity

contrast = mne.evoked.read_evokeds('./S1_discrimination_conds-ave.fif',
                                   condition='contrast_RP', verbose=None)

data = np.array([contrast.data])
data = np.transpose(data, (0, 2, 1))
print('data shape', data.shape)

connectivity, ch_names = find_ch_connectivity(contrast.info, ch_type='mag')
assert connectivity.shape[-1] == len(contrast.info['ch_names'])
```

here is the file: https://www.dropbox.com/s/bae8rjdu068ws14/S1_discrimination_conds-ave.fif?dl=0.

My suggestion would be to make [_compute_ch_connectivity](https://github.com/mne-tools/mne-python/blob/56ce204fda6763458540755988f97c76a69bbcbd/mne/channels/channels.py) public and deprecate the find_ch_connectivity function. The guesswork in the function introduces more problems than solves.

agramfort, 2019-03-21:
for MEG we use fieldtrip neighbors files:

https://github.com/fieldtrip/fieldtrip/tree/master/template/neighbours

I don't see any file for CTF 270.

any idea how to know what CTF system is used, maybe given the channel names?

jasmainak, 2019-03-21:
okay so apparently it's a CTF-275 system with 5 bad channels

agramfort, 2019-03-21:
Arfff

What’s the expected behavior then? I would personally interpolate

mmagnuski, 2019-03-21:
@jasmainak - Fieldtrip neighbours structure gives adjacency information by channel name so you should still be able to use those files. Can you choose channels when reading connectivity in mne?

jasmainak, 2019-03-22:
Yes, you are correct. This is what I realized after posting the issue. I think it's not technically a bug but more a surprising behavior. As people expect the same number of channels in the connectivity matrix as the info. 

Regarding this particular data, the info did not have these 5 channels from the beginning (as far as I know).

agramfort, 2019-03-22:
we could subselect the neighbors structure but it means having holes or
maybe disconnected components...

mmagnuski, 2019-03-22:
@agramfort In the extreme that is possible, but I doubt it would be a common scenario. We could warn when some channels from adjacency are not present in info (and maybe the other way too). Disconnected components are already allowed at the source level (left and right hemi).

agramfort, 2019-03-22:
so warning and subsampling?

>

mmagnuski, 2019-03-22:
Yes - and there is another benefit of accepting the ch_names present in info: if the order of channels is different in fieldtrip neighbours file and the info - they would be sorted in the correct order (I'm not sure it is done now).

agramfort, 2019-03-22:
that would be fine with me.

+1 for PR on this

>