mne-tools/mne-python #6071: Manjaro - Failed building wheel for mayavi

GuillaumeFavelier, 2019-03-21:
On my config, after I follow the [standard installation procedure](https://martinos.org/mne/stable/install_mne_python.html) with `conda env create -f environment.yml`, I got an incomplete install:

`
Failed building wheel for mayavi
Command "/home/guillaume/source/anaconda3/envs/mne/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-gx8ijzup/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-_z6qhh99/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-req-build-gx8ijzup/
`

This is my configuration given by:
```py
import mne
mne.sys_info()
```

```
Platform:      Linux-4.19.28-1-MANJARO-x86_64-with-arch-Manjaro-Linux
Python:        3.6.8 |Anaconda, Inc.| (default, Dec 30 2018, 01:22:34)  [GCC 7.3.0]
Executable:    /home/guillaume/source/anaconda3/envs/mne/bin/python
CPU:           : 4 cores
Memory:        7.7 GB

mne:           0.18.dev0
numpy:         1.16.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
nibabel:       2.3.3
mayavi:        Not found
cupy:          Not found
pandas:        0.24.2
dipy:          Not found
```

I think the origin of the issue is related to my version of `gcc` being too recent:
`gcc --version`
```
gcc (GCC) 8.2.1 20181127
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I got it working by using the version **7.3** of `gcc` in the conda environment:
```py
conda activate mne
conda install gxx_linux-64=7.3
pip install https://api.github.com/repos/enthought/mayavi/zipball/226189a6ad3dc3c01d031ef21d0d0cde554ac851
```

larsoner, 2019-03-21:
That error traceback does not show the real issue. Can you go back to the possibly problematic compiler, checkout that `mayavi` commit, then do:
```
python setup.py install
```
? That will give a full traceback that `pip` otherwise hides. (There might also be a `pip` option to show the compile output but I'm not sure.)

GuillaumeFavelier, 2019-03-22:
I went back to the environment with my default compiler, then executed:
`python setup.py install`
from this specific version of mayavi source code.

I got the following error message:

```
building 'tvtk.array_ext' extension
compiling C sources
C compiler: gcc -pthread -B /home/guillaume/source/anaconda3/envs/mne/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC

creating build/temp.linux-x86_64-3.6/tvtk
creating build/temp.linux-x86_64-3.6/tvtk/src
compile options: '-I/home/guillaume/source/anaconda3/envs/mne/lib/python3.6/site-packages/numpy/core/include -I/home/guillaume/source/anaconda3/envs/mne/include/python3.6m -c'
gcc: tvtk/src/array_ext.c
In file included from /home/guillaume/source/anaconda3/envs/mne/lib/python3.6/site-packages/numpy/core/include/numpy/ndarraytypes.h:1824,
                 from /home/guillaume/source/anaconda3/envs/mne/lib/python3.6/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /home/guillaume/source/anaconda3/envs/mne/lib/python3.6/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from tvtk/src/array_ext.c:223:
/home/guillaume/source/anaconda3/envs/mne/lib/python3.6/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it with " \
  ^~~~~~~
gcc -pthread -shared -B /home/guillaume/source/anaconda3/envs/mne/compiler_compat -L/home/guillaume/source/anaconda3/envs/mne/lib -Wl,-rpath=/home/guillaume/source/anaconda3/envs/mne/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o -o build/lib.linux-x86_64-3.6/tvtk/array_ext.cpython-36m-x86_64-linux-gnu.so
/home/guillaume/source/anaconda3/envs/mne/compiler_compat/ld: build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o: unable to initialize decompress status for section .debug_info
/home/guillaume/source/anaconda3/envs/mne/compiler_compat/ld: build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o: unable to initialize decompress status for section .debug_info
/home/guillaume/source/anaconda3/envs/mne/compiler_compat/ld: build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o: unable to initialize decompress status for section .debug_info
/home/guillaume/source/anaconda3/envs/mne/compiler_compat/ld: build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o: unable to initialize decompress status for section .debug_info
build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
error: Command "gcc -pthread -shared -B /home/guillaume/source/anaconda3/envs/mne/compiler_compat -L/home/guillaume/source/anaconda3/envs/mne/lib -Wl,-rpath=/home/guillaume/source/anaconda3/envs/mne/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o -o build/lib.linux-x86_64-3.6/tvtk/array_ext.cpython-36m-x86_64-linux-gnu.so" failed with exit status 1
```

larsoner, 2019-03-22:
> build/temp.linux-x86_64-3.6/tvtk/src/array_ext.o: file not recognized: file format not recognized

This is a weird error, and does not look specific to Mayavi. Try `make clean` (or delete the `build` directory) then try again

larsoner, 2019-03-22:
(this error probably comes from mixing compiler versions)

GuillaumeFavelier, 2019-03-23:
`make clean` didn't work but I was able to remove the `build` directory before trying again `python setup.py install`. I got the same message again and attached all outputs in [allout.txt](https://github.com/mne-tools/mne-python/files/2999242/allout.txt).

larsoner, 2019-03-23:
Are you on Arch linux? From https://github.com/NicolasHug/Surprise/issues/250 :

https://bbs.archlinux.org/viewtopic.php?id=242682

GuillaumeFavelier, 2019-03-25:
Yes! Manjaro is based on Arch Linux and the error log is similar, thank you.

larsoner, 2019-03-25:
Okay closing since it's an upstream problem

closed 2019-03-25 by larsoner