mne-tools/mne-python #6077: Lack of clear communication in updating reject parameter of mne.Epochs

Alex159, 2019-03-23:
If you still allow the` reject `parameter in the Epochs constructor, but remove its function, then you should give a warning that this update has occurred. I have just wasted a significant amount of time trying to get some code, which works fine on my lab computer (MNE version was installed about a year ago), working on my home computer (with MNE installed today). My data is inherently noisy and my project relates to quantifying this, so I found all my epochs were being dropped on creation of the Epochs object (in _get_data, line 1255 in _epochs.py_) and I  seemingly had no way to turn it off.

I have now just seen an update has occurred, changing `reject` to `reject_by_annotation` but no indication is given to the user and without an error arising due to use of an outdated parameter, it's not clear at all that previous code that is known to work would suddenly fail due to all noisy data being automatically dropped.

I think it's just a case of this communication / warning implementation being overlooked, as I imagine this would affect virtually everyone who creates an Epochs instance and has an issue with non-typical / noisy epochs.

agramfort, 2019-03-25:
@Alex159 sorry to hear this. We try to communicate on any change to the code http://martinos.org/mne/stable/whats_new.html and we have a rigorous policy of deprecations.

you need to realize @Alex159 that most of the work that is done on mne is done by volunteers who hope to have a positive impact of science.

now regarding your problem you must have data with annotations that contain the word BAD which makes them bad data segments.

cbrnr, 2019-03-26:
The `reject` parameter still exists - why do you think it has been removed? Could you share a code snippet that led to your problem?

Alex159, 2019-03-26:
Sure thing. This is code that I was used to writing and works in the older versions of MNE:

```
picks = mne.pick_types(eeg_data.info, eeg=True, meg=False, exclude='bads')  # Pick EEG channels

epochs = mne.Epochs(eeg_data, new_events, event_id=event_id, tmin=-0.1, tmax=0.5,
                   proj=True, picks=picks, baseline=(None,0), preload=True, reject=None)
```

However, if I run this on my noisy data now, I get:

```
8896 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
0 projection items activated
Loading data for 8896 events and 601 original time points ...
/home/alex/anaconda3/lib/python3.7/site-packages/mne/io/base.py:625: RuntimeWarning: invalid value encountered in greater
  start / sfreq)
**IS THIS WHERE?**
8896 bad epochs dropped
/home/alex/anaconda3/lib/python3.7/site-packages/numpy/core/fromnumeric.py:2920: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
/home/alex/anaconda3/lib/python3.7/site-packages/numpy/core/_methods.py:78: RuntimeWarning: invalid value encountered in true_divide
  ret, rcount, out=ret, casting='unsafe', subok=False)
<ipython-input-4-3b496297a1a3>:13: RuntimeWarning: evoked object is empty (based on less than 1 epoch)
  epochs["NOUN"].average().plot(spatial_colors=True, titles=f"", time_unit="ms");
```
The "IS THIS WHERE?" part is just me trying to locate why all my epochs were being rejected, even though I had explicitly set `reject` to be `None`. That happens in `_get_data` of epochs.py, where `self._bad_dropped` is set to `True` even though I don't want this to happen. That's how I spotted this is where the rejection was happening. 

I imagine it has something to do with the annotations then. I don't manually annotate anything yet (nor am I aware of a procedure that would do this implicitly in my code), so I think a new behaviour in how this works might trigger mass rejection and mean that the only way to save my epochs is now setting `reject_by_annotations `to `False `(which is a new parameter that has come in a later version of MNE).

cbrnr, 2019-03-26:
Thanks, this behavior looks like a bug and was certainly not intended. You are right that it might have to do with our recent switch to annotations - @massich do you have time to take a look?

agramfort, 2019-03-26:
I can take a look

can you share a full code snippet with the data to replicate?

>

Alex159, 2019-03-26:
> I can take a look can you share a full code snippet with the data to replicate?
> […](#)

Hi, Alexandre. 
Thanks for taking a look. Here is the code snippet below.
I have also cropped 30 seconds of the data and attached at the end of this reply (~ 7.5 MB).

```
import mne
print(mne.__version__)

# Load the cropped EEG data
eeg_data_cropped = mne.io.read_raw_fif('eeg_data_cropped_raw.fif');

# Recover events from cropped EEG
eeg_events = mne.find_events(eeg_data_cropped, stim_channel='STI 014',
                             shortest_event=1);

picks = mne.pick_types(eeg_data_cropped.info, eeg=True, meg=False, exclude='bads');

epochs = mne.Epochs(eeg_data_cropped, eeg_events, event_id={'fixation':252}, tmin=-0.1, tmax=0.4,
                    proj=True, picks=picks, baseline=None, preload=True, reject=None)
```
And the accompanying output:
```
0.17.1
Opening raw data file C:\Users\...\eeg_data_cropped_raw.fif...
    Range : 460000 ... 490000 =    460.000 ...   490.000 secs
Ready.
291 events found
Event IDs: [  1   2   3   4   5   6   7   8   9  10  11  12  13  14  15  16  17  18
  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35  36
  37  38  39  40  41  42  43  44  45  46  47  48  49  51  52  53  54  55
  56  57  58  59  60  61  62  63  64  65  66  67  68  69  70  71 250 252]
188 matching events found
No baseline correction applied
Not setting metadata
0 projection items activated
Loading data for 188 events and 501 original time points ...
188 bad epochs dropped
```

As you can see, `reject` is set to `None` but all the epochs that were found were then dropped.

[eeg_data_cropped_raw.zip](https://github.com/mne-tools/mne-python/files/3008565/eeg_data_cropped_raw.zip)

agramfort, 2019-03-27:
here is the problem:

In [4]: raw.annotations.description[raw.annotations.description == 'bad_ET']
Out[4]:
array(['bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET', 'bad_ET',
       'bad_ET'], dtype='<U12')

In [5]: raw.annotations.duration[raw.annotations.description == 'bad_ET']
Out[5]:
array([20.19300079, 22.07999992, 24.39500046, 28.77300072, 30.00099945,
       30.00099945, 30.00099945, 30.00099945, 30.00099945, 30.00099945,
       30.00099945, 30.00099945, 30.00099945, 30.00099945, 30.00099945,
       30.00099945, 30.00099945, 30.00099945, 30.00099945, 30.00099945,
       30.00099945, 30.00099945, 30.00099945, 30.00099945, 30.00099945,
       30.00099945, 30.00099945, 30.00099945, 30.00099945, 30.00099945,
       30.00099945, 30.00099945, 30.00099945, 30.00099945, 30.00099945,
       30.00099945, 28.7689991 , 28.38899994, 19.24699974, 18.86599922,
        7.13299942,  6.78199959,  1.35199928])

you have bad segments that last the duration of the chunk of data.

I don't know where these annotations come from but to get the old behavior
you can do:

raw = mne.io.read_raw_fif('eeg_data_cropped_raw.fif');
raw.set_annotations(None)

Alex159, 2019-04-23:
Hi Alexandre,

That solution worked for me for a while, but now the issue is back again. I have confirmed that the annotations have been removed, by including your suggestion of applying the `set_annotations(None) `immediately after loading the file. 

What happens now is that an SSP operator is being invoked, where it wasn't before, and that is causing rejection of 80% of epochs.

```
print(raw_ica.annotations)
epochs = mne.Epochs(raw_ica, events_new, pos2code, baseline=(-0.2, 0), tmin=-0.2, tmax=0.3,
                    proj=True, picks=picks_eeg, preload=True,
                    reject=None, reject_by_annotation=None)
```
And what I see is:

```
<Annotations  |  0 segments , orig_time : 2019-04-15 15:26:10.299999>
5770 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
Created an SSP operator (subspace dimension = 1)
1 projection items activated
Loading data for 5770 events and 501 original time points ...
5213 bad epochs dropped
```

Having set the annotations to `None`, confirming there is nothing there (top line of output) and setting both `reject `and `reject_by_annotation` to be `None`, the epochs I want to keep are being dropped by a rejection procedure that I do not wish to occur.

Do you have a tip on what might be a solution to stopping these epochs from being removed?

Many thanks.
I really do appreciate the MNE team's work on this library.

Alex

larsoner, 2019-04-23:
`proj=False` in the epochs constructor will prevent it from being applied. But a better solution might be to remove the projectors in the first place with `raw.del_proj()` earlier in your pipeline

Alex159, 2019-04-23:
Hi larsoner,

Thank you for the quick reply. I see a difference in the ERPs when I toggle the `proj `parameter to on / off but it doesn't actually stop the epochs from being dropped. I'm afraid even explicitly deleting the projection has no effect on the dropping of epochs:

```
print(raw_ica.annotations)

raw_ica.del_proj()

epochs = mne.Epochs(raw_ica, events_new, pos2code, baseline=(-0.2, 0), tmin=-0.2, tmax=0.3,
                    proj=False, picks=picks_eeg, preload=True, reject=None, reject_by_annotation=None)
```

```
<Annotations  |  0 segments , orig_time : 2019-04-15 15:26:10.299999>
5770 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
Loading data for 5770 events and 501 original time points ...
5213 bad epochs dropped
```
As I mentioned, above I have proj=False, but exactly same output whether it is set to True or False (though I can confirm the plotted results do differ, so the effect still takes place, but has no bearing on the rejection issue, unfortunately).

mmagnuski, 2019-05-04:
I cannot replicate on the provided data (eeg_data_cropped_raw.zip).
I add evenly spaced events this way:
```python
times = np.arange(1, 30.)
smp = times * raw.info['sfreq'] + raw.first_samp
events = np.zeros((len(smp), 3), dtype='int')
events[:, 0] = smp.astype('int')
events[:, -1] = 1
```
and then run:
```python
epochs = mne.Epochs(raw, events, baseline=(-0.2, 0), tmin=-0.2, tmax=0.3,
                    proj=False, picks='eeg', preload=True, reject=None,
                    reject_by_annotation=None)
```

nothing gets rejected.

larsoner, 2019-08-21:
This appears to be a problem with custom code having to do with `first_samp`, I recommend using `make_fixed_length_events` instead. No changes needed here so I'll close, thanks for the report

closed 2019-08-21 by larsoner

Christonikos, 2021-02-18:
Dear developers, 

I am facing the exact same problem: all of my epochs are being dropped automatically, without me explicitly asking for it. 

I tried to implement all of your suggestions, including these taken from issue #5992, but to no success. 

I am attaching to this thread:

- A standalone script that replicates the issue. 
- A small snippet from my recording (30 seconds).
- The events and the event_id files. 
- The specs of my environment. 

[run_01-eve.fif.zip](https://github.com/mne-tools/mne-python/files/5999322/run_01-eve.fif.zip)
[replicate_issue.zip](https://github.com/mne-tools/mne-python/files/5999326/replicate_issue.zip)

Thanks so much for your time. 
Christos

cbrnr, 2021-02-18:
@Christonikos you forgot to attach `run_01_sss_raw.fif` (the raw data).

Christonikos, 2021-02-18:
Hi @cbrnr, my bad. You can find the data [here](https://we.tl/t-Vag8QWTOWR).

cbrnr, 2021-02-18:
I confirm that all epochs are dropped (`epochs.drop_log` says all of them are `TOO_SHORT`).

I suspect that this has something to do with event positions starting at sample 770624:

```
>>> events
array([[ 770624,       0,     550],
       [ 771239,       0,      54],
       [ 771305,       0,      55],
       ...,
       [1487464,       0,     213],
       [1487530,       0,     218],
       [1488797,       0,     219]])
```

The continuous signal has length 371000:

```
>>> raw
<Raw | run_01_sss_raw.fif, 390 x 371000 (742.0 s), ~1.08 GB, data loaded>
```

But `raw.first_time` and `raw.first_samp` seem to be correctly set:

```
>>> raw.first_samp, raw.first_time
(380500, 761.0)
```

Not sure, but shouldn't `mne.Epochs` take care of this? @larsoner @hoechenberger?

Christonikos, 2021-02-18:
Hi, we can close the thread. It was a stupid issue, I just hope that this can serve as a reference to other users as well. 

I had resampled my data after extracting the events. Therefore, the sampling rates did not match and that is why the epochs were rejected as "TOO_SHORT". 

I would like to thank @cbrnr for his time, and I am sorry that I bothered you with something so stupid. 
Christos

cbrnr, 2021-02-18:
Glad to hear you figured out the problem!

svagatt, 2022-01-24:
Hi,
Unfortunately, I am going through a similar problem, and I have tried to do the things suggested above, the raw.first_samp() and raw.first_time() are 0 and 0. I am trying to extract a single epoch from raw data that's approximately 4.4s. When I try to get the epochs, my 1 epoch with the event that I need is getting dropped as. abad epoch although I set both the rejects to None.
`    epoch = mne.Epochs(raw, events, event_dict, -0.1, 2.0, (None, 0.0), preload=True, reject=None, reject_by_annotation=None)
`
this is the error I get
`0 0.0
[[1505, 0, 39], [1505, 0, 99]]
1 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
0 projection items activated
Loading data for 1 events and 1051 original time points ...
1 bad epochs dropped
`
I have attached the file and the data
[epoch_error.zip](https://github.com/mne-tools/mne-python/files/7928893/epoch_error.zip)