mne-tools/mne-python #6085: ENH: Add minimum-norm based model for virtual sensors

larsoner, 2019-03-26:
The `map_data` functions should become public somehow for mapping data between runs.

I think we can do it pretty easily actually by incorporating it into `maxwell_filter` (you get many things for free this way) but Alex does not like it yet. For discussion at the sprint.

jasmainak, 2019-03-26:
maybe something like: https://gist.github.com/jasmainak/756cf02dce82ab2dc5c2c69722dd13d1

larsoner, 2019-03-27:
Also as suggested by @jjnurminen `_map_meg_channels` could also use multipole moments instead of minimum norm. I think ideally one should be able to use either transformation approach (minimum norm or multipole moments) in both `maxwell_filter` and in whatever ends up being the public version of `_map_meg_channels`.

See also https://github.com/mne-tools/mne-python/pull/4491 where @kingjr wanted things to be faster -- the multipole moment method could end up being faster, but either way we can tackle speed as part of this issue.

larsoner, 2023-10-01:
I think we have this sort of thing with https://github.com/mne-tools/mne-python/pull/7366 and epochs/evoked.as_type

FWIW the transformation is linear so if you want it to be really fast, do it once with an `EvokedArray(np.eye(info["nchan"]), info)` or so and then do the `matmul` to data as needed

closed 2023-10-01 by larsoner