mne-tools/mne-python #6086: ENH: Add distributed parallelism / scaling

larsoner, 2019-03-26:
e.g. using Kubernetes/dask/joblib. Need to quantify scaling performance.

SherazKhan, 2021-01-06:
@larsoner continuing our discussion on using Dask as the backend for parallel jobs. I was wondering, maybe we can start by having an option to specify which backend to use for joblib?

```
from dask.distributed import Client
import joblib
import os.path as op
import mne
from mne.datasets import sample

n_workers = 4
threads_per_worker = 1
client = Client(n_workers=n_workers, threads_per_worker=threads_per_worker)             # create local cluster

data_path = sample.data_path()
raw_fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(raw_fname, preload=True)

with joblib.parallel_backend('dask'): # using dask backend
    raw.filter(0.1, 2, n_jobs=n_workers)
```

```
[Parallel(n_jobs=4)]: Using backend DaskDistributedBackend with 4 concurrent workers.
[Parallel(n_jobs=4)]: Done  14 tasks      | elapsed:    1.9s
[Parallel(n_jobs=4)]: Done 366 out of 366 | elapsed:    5.7s finished
```

agramfort, 2021-01-06:
I agree it's the best way to go presently.

@hoechenberger I think it could help a lot with https://github.com/mne-tools/mne-study-template/

larsoner, 2021-01-06:
This sounds reasonable to me, want to give it a try @SherazKhan ?

SherazKhan, 2021-01-06:
@larsoner definitely 

I was thinking the easiest way is to expose joblib.Parallel, backend argument in mne-python parallel_func function. Something like this:

```
def parallel_func(func, n_jobs, max_nbytes='auto', pre_dispatch='n_jobs',
                  total=None, prefer=None, verbose=None, backend=None):

    joblib.Parallel(n_jobs=None, backend=backend)

```

larsoner, 2021-01-06:
Just put `backend=None` before `verbose=None` (that one always stays at the end) and +1 from me!

larsoner, 2021-01-06:
... wait, do we even need a `backend=None` arg? We currently only expose `n_jobs` so even without a `backend=None` arg it seems like the code you wrote above should already work:
```
with joblib.parallel_backend('dask'): # using dask backend
    raw.filter(0.1, 2, n_jobs=n_workers)
```
This doesn't seem to require an API change from MNE, does it?

SherazKhan, 2021-01-06:
No, it does not requires changing API,  `with joblib.parallel_backend('dask')` overwrites default _loky_ backend.

However, I think API change will make integration tighter.

agramfort, 2021-01-07:
not everything will make sense in terms of usage with data

filtering channels in parallel in dask over different machines. I don't get
the point.
Unless you use dask on one machine where loky would do as well

To me it really makes sense across subjects

>

SherazKhan, 2021-01-07:
@agramfort I completely agree, my use case is specific since I am processing data on the cloud, with a weak master node and strong serverless on-demand workers.

It makes more sense across subjects.

larsoner, 2021-01-07:
Should we perhaps focus on implementing this at the level of `mne-study-template`, then?

agramfort, 2021-01-07:
> Should we perhaps focus on implementing this at the level of
mne-study-template, then?

I would say yes

cc @hoechenberger

>

SherazKhan, 2021-01-07:
@larsoner yes, I will also add some example scripts on processing big data on major cloud services. 

This also goes back to our discussion of the official mne docker, which can be used to deploy computing engines on the worker nodes. 

Instead of making separate images for each cloud provider, we will use this docker agnostically.
Maybe we can generate this docker image as a part of Github Actions and push it to Docker Hub.

hoechenberger, 2021-01-07:
So my idea was to pack the Study Template into a Docker container anyway, and we could simply spin up multiple instances in parallel, does that sound good? Should be almost trivial, now that we have our `run.py` script already

hoechenberger, 2021-01-07:
Distribution across machines could happen via Kubernetes

SherazKhan, 2021-01-07:
@hoechenberger MNE's Docker image can be based on the [official freesurfer docker image](https://hub.docker.com/r/freesurfer/freesurfer)

hoechenberger, 2021-01-07:
@SherazKhan Thanks, I already have a bunch of Python Dockerfiles around here :) But will have a look at that one too. I'd definitely want to keep FS in a separate image, though
Which brings me back to an old issue I started working on over a year ago and then stopped because I moved to Paris 😅 
https://github.com/mne-tools/mne-python/issues/7112

Will pick up working on this again soon. Still have some other, more urgent things on my plate. Do you happen to have any time pressure, @SherazKhan?

SherazKhan, 2021-01-07:
> Will pick up working on this again soon. Still have some other, more urgent things on my plate. Do you happen to have any time pressure, @SherazKhan?

@hoechenberger, Since distributed computing on the cloud, was in Aim 2 using Dask of the MNE grant, maybe it will be good to show some examples before the next progress report (in 3 months).  Please let me know, how I can contribute?

jasmainak, 2021-01-08:
I feel before jotting down an API, we should try to come up with use-cases. When do you need distributed parallelism? If you parallelize across subjects, how do you manage memory? Parallelism multiplies the memory. I imagine different parts of the pipeline may need different levels of parallelism because of this. Just throwing ideas ... trying on some datasets may help guide the usecases

agramfort, 2021-01-08:
use case:

process cam-can in 1h or less :)

>

larsoner, 2022-02-01:
Closed by https://github.com/mne-tools/mne-bids-pipeline/pull/472 !

closed 2022-02-01 by larsoner