mne-tools/mne-python #6087: mne.set_config(None) doesn't work as described

drammock, 2019-03-26:
The [docs](https://mne-tools.github.io/dev/generated/mne.set_config.html) for `mne.set_config` say:

```
mne.set_config(key, value, home_dir=None, set_env=True)
Parameters:
key : str | None
    The preference key to set. If None, a tuple of the valid keys is returned, and value and home_dir are ignored.
```

however, if `key` is `None` then `value` is *not* ignored:

```
(mnedev) [~] $ ipython
In [1]: import mne                                                                                                                    
In [2]: mne.set_config(None)                                                                                                          
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-2-26697f64a5bd> in <module>
----> 1 mne.set_config(None)

TypeError: set_config() missing 1 required positional argument: 'value'
In [3]: mne.sys_info()                                                                                                                
Platform:      Linux-4.15.0-46-generic-x86_64-with-debian-buster-sid
Python:        3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 19:16:44)  [GCC 7.3.0]
Executable:    /opt/miniconda3/envs/mnedev/bin/python
CPU:           x86_64: 8 cores
Memory:        7.6 GB

mne:           0.18.dev0
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    3.0.2 {backend=Qt5Agg}

sklearn:       0.20.2
nibabel:       2.3.1
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.23.4
dipy:          0.15.0
```

larsoner, 2019-03-26:
It should say `If `key` and `value` are both None, ...`

cbrnr, 2019-03-27:
Alternatively, it would probably be more intuitive if `mne.set.config()` returned the valid keys (i.e. by default the function should use `key=None, value=None`).

massich, 2019-03-27:
+1 to set them both to `None` by default and use `mne.set_config()` to get the answer. (even if `set_config` to mean **get** me the confing might seem weird)

larsoner, 2019-03-27:
It does not get the config, just the valid keys that you can use to set the configuration. get_config() gets the current configuration (keys and values).

PR #6175 (2019-04-23): MRG: move "show all config keys" functionality from set_config into get_config
closes #6087 

@massich @larsoner @cbrnr
closed 2019-04-23 by larsoner