mne-tools/mne-python #6091: problems when loading edf file

olipa, 2019-03-27:
#### Describe the bug
I get some unexpected behaviors when loading the following file:

https://www.dropbox.com/s/nb5sfl1rxsx887p/test_annon_mne.edf?dl=0

unexpected behaviours:

* I lose the `EOG M1` channel ( when `stim_channel='auto'`)
* I lose the miliseconds in `raw.info['meas_date']` (despite being in the file)
* when I do `mne.find_events(raw)` (when `stim_channel` is present) I get errors

#### Steps and/or code to reproduce
```Python
In [6]: with pytest.warns(RuntimeWarning, match='overlapping'): 
   ...:     raw = mne.io.read_raw_edf('test_annon_mne.edf', preload=True) 
   ...:                                                                                                                                                                     
Extracting EDF parameters from test_annon_mne.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 3579391  =      0.000 ... 13981.996 secs...
Used Annotations descriptions: ['Sleep stage W', 'EMG : EMG haut', 'EOG : REM', 'EEG : Delta', 'EMG : Non valide', 'EOG : Non valide', 'EMG : EMG mÃ©dium', 'EEG : Alpha', 'EMG : Mouvement', 'EMG : EMG trÃ¨s bas', 'EEG : Fuseau', 'Sleep stage N3', 'Non valide', 'VidÃ©o', 'Note : rox', 'Note : * * Join position * * *', 'Sleep stage R', 'Sleep stage N1']

In [7]: raw.info['ch_names']                                                                                                                                                
Out[7]: 
['EDF Annotations',
 'EEG F3',
 'EEG F4',
 'EEG C3',
 'EEG C4',
 'EEG O1',
 'EEG O2',
 'EMG EMG Chin',
 'ECG',
 'EOG E1',
 'EOG E2',
 'EOG M2',
 'STI 014']

In [8]: raw.info                                                                                                                                                            
Out[8]: 
<Info | 16 non-empty fields
    bads : list | 0 items
    ch_names : list | EDF Annotations, EEG F3, EEG F4, EEG C3, EEG C4, EEG O1, ...
    chs : list | 13 items (STIM: 2, EEG: 11)
    comps : list | 0 items
    custom_ref_applied : bool | False
    dev_head_t : Transform | 3 items
    events : list | 0 items
    highpass : float | 0.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 128.0 Hz
    meas_date : tuple | 2031-07-10 08:38:48 GMT
    nchan : int | 13
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 256.0 Hz
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    description : NoneType
    dev_ctf_t : NoneType
    dig : NoneType
    experimenter : NoneType
    file_id : NoneType
    gantry_angle : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    meas_id : NoneType
    proj_id : NoneType
    proj_name : NoneType
    subject_info : NoneType
    xplotter_layout : NoneType
>

In [9]: raw.annotations                                                                                                                                                     
Out[9]: <Annotations  |  1407 segments : eeg (580), emg (441), eog (305)..., orig_time : 2031-07-10 08:38:48>

In [10]: mne.find_events(raw)                                                                                                                                               
Trigger channel has a non-zero initial value of 20 (consider using initial_event=True to detect this event)
1065 events found
Event IDs: [ 2  3  4  5  6  7  8  9 10 11 14 16 18 19 20 21]
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-10-aa6cf1cc89fa> in <module>
----> 1 mne.find_events(raw)

~/miniconda3/envs/mne/lib/python3.6/site-packages/mne/event.py in find_events(raw, stim_channel, output, consecutive, min_duration, shortest_event, mask, uint_cast, mask_type, initial_event, verbose)

~/miniconda3/envs/mne/lib/python3.6/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    950         with use_log_level(verbose_level):
    951             return function(*args, **kwargs)
--> 952     return function(*args, **kwargs)
    953 
    954 

~/miniconda3/envs/mne/lib/python3.6/site-packages/mne/event.py in find_events(raw, stim_channel, output, consecutive, min_duration, shortest_event, mask, uint_cast, mask_type, initial_event, verbose)
    705                              "e.g. x / raw.info['sfreq']. Where x = 1 sample "
    706                              "shorter than the shortest event "
--> 707                              "length." % (n_short_events))
    708 
    709         events_list.append(events)

ValueError: You have 1 events shorter than the shortest_event. These are very unusual and you may want to set min_duration to a larger value e.g. x / raw.info['sfreq']. Where x = 1 sample shorter than the shortest event length.

In [12]: raw = mne.io.read_raw_edf('test_annon_mne.edf', stim_channel=False, preload=True) 
Extracting EDF parameters from test_annon_mne.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<ipython-input-12-0867d9c72914>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
  raw = mne.io.read_raw_edf('test_annon_mne.edf', stim_channel=False, preload=True)
Reading 0 ... 3579391  =      0.000 ... 13981.996 secs...

In [13]: raw.info['ch_names']                                                                                                                                               
Out[13]: 
['EDF Annotations',
 'EEG F3',
 'EEG F4',
 'EEG C3',
 'EEG C4',
 'EEG O1',
 'EEG O2',
 'EMG EMG Chin',
 'ECG',
 'EOG E1',
 'EOG E2',
 'EOG M2',
 'EOG M1']

```

#### Additional information
Paste the output of `mne.sys_info()` here.
```py
In [14]: mne.sys_info()                                                                                                                                                     

Platform:      Linux-4.4.0-142-generic-x86_64-with-debian-stretch-sid
Python:        3.6.8 |Anaconda, Inc.| (default, Dec 30 2018, 01:22:34)  [GCC 7.3.0]
Executable:    /home/pallanca/miniconda3/envs/mne/bin/python
CPU:           x86_64: 20 cores
Memory:        250.5 GB

mne:           0.17.1
numpy:         1.16.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
nibabel:       2.3.3
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.24.2
dipy:          0.16.0
```

cc: @massich, @cbrnr

massich, 2019-03-27:
Some of the issues you have reported they are not present in `0.18.dev`
(but its true that we do not recover the ms properly)

```py
import mne
import os.path as op
from mne import __file__ as _mne_file

fname = op.join(op.dirname(_mne_file), '..', 'sandbox',
                'data', 'test_annon_mne.edf')

if __name__ == "__main__":
    raw = mne.io.read_raw_edf(fname)
    print(raw.info['ch_names'])  # no 'EDF Annotations' or 'STI 014'
    print('date (s, ms): ', raw.info['meas_date'])  # (1941439128, 483584)

    # mne.find_events(raw)  fails because there is 'STI 014' no more since 0.18
    events, events_id = mne.events_from_annotations(raw)
    
```

```sh
['EEG F3', 'EEG F4', 'EEG C3', 'EEG C4', 'EEG O1', 'EEG O2', 'EMG EMG Chin', 'ECG', 'EOG E1', 'EOG E2', 'EOG M2', 'EOG M1']
date (s, ms):  (1941439128, 0)
```

To write an equivalent code in `0.17.1` it should be something like this:

```py
import warnings

with warnings.catch_warnings(record=True) as recwarn:
    raw = mne.io.read_raw_edf(fname, stim_channel=False, preload=True)
raw = raw.drop_channels(['EDF Annotations'])
raw.set_annotations(mne.read_annotations(fname))

events, events_id = mne.events_from_annotations(raw)
```

massich, 2019-03-27:
But this file is buggy in `0.17` check if you recover the expected `events` and maybe you just want to use `0.18.dev`

cbrnr, 2019-03-28:
I can confirm that everything works as expected with `0.18.dev`. Regarding the time stamp, EDF does not record microseconds, so MNE just sets this value to zero. BTW, your recording date is July 10 1931 - are you sure this is correct or is this one of Hans Berger's EEG recordings :smile:?

cbrnr, 2019-03-28:
@massich FWIW I think the second element contains the additional microseconds, not milliseconds, at least according to the [docs](https://martinos.org/mne/stable/generated/mne.Info.html).

olipa, 2019-03-28:
Hi

Thank you , i will probably try the 0.18dev : )
Yes we set the recording to the ancient time for anonymisation : ) and I
think we are in 2031 on the edf file !

By the way , if you invite me, I am planning to come to the "spring week" ,
but  obviously I will be "underefficient" on MNE , i hope it's not a
problem...
Best

Olivier

cbrnr, 2019-03-28:
Alright, so I guess problem solved then. Feel free to open a new issue if you encounter more problems.

closed 2019-03-28 by cbrnr

larsoner, 2019-03-28:
Do we correctly extract the `ms` offset (which would end up in `info['meas_date'][1]` multiplied by a factor of 1000), or is `info['meas_date'][1]` always `0`?

larsoner, 2019-03-28:
@olipa what should the `ms` value be for this file you uploaded?

cbrnr, 2019-03-28:
I've only ever seen values of `0`, but maybe because EDF/BDF don't store ms/us.

larsoner, 2019-03-28:
Looking at how we convert the meas date, I don't see where `ms` could even be stored in the EDF header. We extract from day/month/year and hour/min/sec fields that are each 8 char long like:
```
10.07.31
08.38.48
```
then the next `fid.read` is the number of header bytes. So really nowhere to store with sub-second-level precision that I can see.

cbrnr, 2019-03-28:
Yes, EDF doesn't store ms, so I think everything works as expected.

massich, 2019-03-28:
> We extract from day/month/year and hour/min/sec fields that are each 8 char long

yep, my fault. `3584` is `header_nbytes`. 

all good.

olipa, 2019-03-28:
Hi

sorry for the delay , indeed i could check  today that even in the original
file the starting point is :

08.38.48.000000, so it could not be ms

So no more precision

thanks

viyyer, 2021-02-27:
I can still reproduce this bug in mne-0.22.0

agramfort, 2021-02-28:
with what file?

In [1]: import mne
raw = mne.io.read
In [2]: raw = mne.io.read_raw('test_annon_mne.edf')
<ipython-input-2-2561301fcce1>:1: RuntimeWarning: Omitted 2391
annotation(s) that were outside data range.
  raw = mne.io.read_raw('test_annon_mne.edf')

works

>